embedded-office/canopen-stack #5: [Question]: Sync received application callbacks?

jcatsong, 2020-07-13:
**Description**

This is not a bug, rather a question about using the stack, I didnt find anywhere else to post questions so I hope this is ok.  I'm trying to determine if its possible to have an application callback occur once a sync message has been received by the stack and processed?  The goal would be to allow the application to be notified when a sync has been received so it can read updated object dictionary values/PDO's/etc.

I see there is an application hook for COPdoReceive(), but this includes the bare CO_IF_FRM * received by the stack, and doesn't include sync information.

Thank you for the nice stack btw, its very easy to use and the documentation is nice.

**Environment (If Necessary)**
- CANopen Stack: 3.3.0

michael-hillmann, 2020-07-14:
Hi, you are welcome to ask your questions at this place.

If I understand you right, you need a callback function that is called after a synchronized RPDO is processed?
Unfortunately, there is no callback today.

We could add a callback in file co_sync.c in function [COSyncHandler()](https://github.com/embedded-office/canopen-stack/blob/67f7b1c9f5695f31fdcf2d2e384da0465d5ebb65/canopen/source/co_sync.c#L61).

May I ask, what you want to achieve with the callback function?

jcatsong, 2020-07-14:
Hi Michael,

Thanks for the quick reply.  Yes as I understand it I'd like an application callback that occurs after a synchronized RPDO is processed and written to the OD.  I see the COSyncHandler() function you mentioned and I imagine a callback would be fit right at the end of the function call and then need to be implemented by the user.

Regarding what I'm hoping to achieve, I'd like to allow my application firmware to be notified when the CANOpen master updates a value in the object dictionary via synchronized PDO (potentially the callback would just signal a semaphore, place a message in a queue, etc).  This is my first application with CANOpen so maybe there's another way to achieve this.  I dont believe polling the object dictionary would work as the master could send the same value via synchronized PDO and the node would still need to act on it (in our application).

michael-hillmann, 2020-07-14:
Ok, let us prepare a callback for the next version like:

~~~c
    for (i = 0; i < CO_RPDO_N; i++) {
        if (sync->RPdo[i] != 0) {
            CORPdoWrite(sync->RPdo[i], &sync->RFrm[i]);
            COSyncUpdateCallback(sync->RPdo[i]);
        }
    }
~~~

With this enhancement, you can notify the RPDO identifier in a queue (or just set a semaphore). I think the CAN frame is not needed in this callback function, because we should always use the data out of the object dictionary. Will this change cover your needs?

closed 2020-07-14 by michael-hillmann

jcatsong, 2020-07-14:
Hi Michael,

This sound like exactly what we need, thank you for the quick response and quick suggestion.  I'll add your suggested changes locally and let you know if I have any issues, but I think it will work for our needs.

Agreed we should use the object dictionary data rather than the CAN frame directly here.  Thanks again!