embedded-office/canopen-stack #9: Missing object dictionary entries in quickstart example

cesarvandevelde, 2020-09-20:
**Description**

First of all, thank you for releasing this high-quality stack under an open source license! I've worked with the CanFestival stack in the past, but it is not actively maintained and has some strange quirks.

While experimenting with this stack, I bumped into a couple of documentation issues. The steps described in the [quickstart guide](https://canopen-stack.org/docs/start/quickstart) do not lead to a minimal working example of the CANopen stack. In the example, only the first subindex of the identity object (1018h) is given, but the LSS module requires all 4 subindices. Similarly, the Sync COB-ID (1005h) is missing from the example, but is required by the sync subsystem. Finally, I think it would be a good idea to show explicitly how object dictionary variables (e.g. `Obj1001_00_08`) need to be defined. 

**Steps to Reproduce**

1. Create a node with an object dictionary as given by the quickstart guide

```C
#include "co_core.h"

/* define the static object dictionary */
const CO_OBJ ClockOD[] = {
    {CO_KEY(0x1000, 0, CO_UNSIGNED32|CO_OBJ_D__R_), 0, 0},
    {CO_KEY(0x1001, 0, CO_UNSIGNED8 |CO_OBJ____R_), 0, &Obj1001_00_08},
    {CO_KEY(0x1017, 0, CO_UNSIGNED16|CO_OBJ_D__R_), 0, 0},
    {CO_KEY(0x1018, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, 1},
    {CO_KEY(0x1018, 1, CO_UNSIGNED32|CO_OBJ_D__R_), 0, 0},

    {CO_KEY(0x1200, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, 2},
    {CO_KEY(0x1200, 1, CO_UNSIGNED32|CO_OBJ_DN_R_), 0, 0x600},
    {CO_KEY(0x1200, 2, CO_UNSIGNED32|CO_OBJ_DN_R_), 0, 0x580},

    {CO_KEY(0x1800, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, 2},
    {CO_KEY(0x1800, 1, CO_UNSIGNED32|CO_OBJ_D__R_), 0, 0x40000180},
    {CO_KEY(0x1800, 2, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, 254},

    {CO_KEY(0x1A00, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, 3},
    {CO_KEY(0x1A00, 1, CO_UNSIGNED32|CO_OBJ_D__R_), 0, 0x21000120},
    {CO_KEY(0x1A00, 2, CO_UNSIGNED32|CO_OBJ_D__R_), 0, 0x21000208},
    {CO_KEY(0x1A00, 3, CO_UNSIGNED32|CO_OBJ_D__R_), 0, 0x21000308},

    {CO_KEY(0x2100, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, 3},
    {CO_KEY(0x2100, 1, CO_UNSIGNED32|CO_OBJ___PR_), 0, &Obj2100_01_20},
    {CO_KEY(0x2100, 2, CO_UNSIGNED8 |CO_OBJ___PR_), 0, &Obj2100_02_08},
    {CO_KEY(0x2100, 3, CO_UNSIGNED8 |CO_OBJ___PR_), 0, &Obj2100_03_08},  
};
/* set number of object entries */
const uint16_t ClockODLen = sizeof(ClockOD)/sizeof(CO_OBJ);
```

2. Compile and start the example. Observe that the application crashes and `Clk->Error` is set to `CO_ERR_CFG_1018` inside `COLssInit()`.

3. Add extra subindices for product code, revision ID, and serial:

```C
    {CO_KEY(0x1018, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, 4},
    {CO_KEY(0x1018, 1, CO_UNSIGNED32|CO_OBJ_D__R_), 0, 0},
    {CO_KEY(0x1018, 2, CO_UNSIGNED32|CO_OBJ_D__R_), 0, 0},
    {CO_KEY(0x1018, 3, CO_UNSIGNED32|CO_OBJ_D__R_), 0, 0},
    {CO_KEY(0x1018, 4, CO_UNSIGNED32|CO_OBJ_D__R_), 0, 0},
```

4. Observe that the application still crashes. `Clk->Error` is now set to `CO_ERR_CFG_1005_0` inside `COSyncInit()`.

5. Add the missing index:

```C
   {CO_KEY(0x1005, 0, CO_UNSIGNED32|CO_OBJ____RW), 0, (uintptr_t)&Obj1005_00_32},
```

6. The example should now run.

**Expected behavior**

Ideally, the quickstart guide should build into a working example application.

**Environment (If Necessary)**

- Target: STM32F103
- Cross-Compiler: GNU Arm embedded toolchain 9-2020-q2-update
- CANopen Stack: 4.0.1

michael-hillmann, 2020-09-21:
Thank you for your time to explain this issue. You are right - we should improve this first experience.

In general, we have two possible solutions:
1. stay with forcing the object entries for LSS and SYNC and improve the quickstart example.
2. stay with the quickstart example and allow the absence of these object entries in case that LSS and SYNC are not required.

What is your opinion, which way is preferred?

michael-hillmann, 2020-09-21:
In my opinion, it is okay to force these object entries (5 x 32byte = 160byte FLASH consumption) and your suggested improvements in the quickstart example documentation.

closed 2020-09-21 by michael-hillmann

cesarvandevelde, 2020-09-22:
Hello @michael-hillmann, thank you for your quick response! I agree with you, making these entries mandatory is probably the best solution. The 160 byte size impact is justified in my opinion, seeing that CANopen devices tend to use at least "medium-size" microcontrollers (e.g. ARM M3/M4 devices) with sufficient amounts of flash / RAM. With that in mind, I think it's best to see the LSS module as functionality you get "for free" with minimal effort, rather than a module you can enable/disable.

On a side note: I ran across some other points that I think merit discussion, but I did not want to spam you with issues. Of course, I'm also willing to contribute pull requests, if you are open to that...

michael-hillmann, 2020-09-22:
Hi @cesarvandevelde, you are welcome to contribute in any form. With issues we will get a good documentation for all users, what is changed. With a pull request we are more efficient (I guess). I'm open to any help to get this project a useful library!

_As a small side note: I'm working with Perforce on my daily work - so I'm in the training phase with github and the git work flow ;-) I will give my best to did it right._