embedded-office/canopen-stack #13: Make CO_TMR_TICKS_PER_SEC configurable

cesarvandevelde, 2020-09-24:
**Description**

`CO_TMR_TICKS_PER_SEC` is currently hardcoded to a value of 100 inside `co_tmr.h`. It might be useful to make this value configurable, seeing that many CANopen communication parameters are more fine-grained than 10ms. Examples include cycle period (in us), TPDO inhibit time (in multiples of 100us), and producer heartbeat time (in ms).

**Proposed solution**

The simplest solution would be to move `CO_TMR_TICKS_PER_SEC` from `co_tmr.h` to `co_cfg.h`.

Optionally, the original definition inside `co_tmr.h` could be wrapped so that a fall-back value is used if `CO_TMR_TICKS_PER_SEC` is missing, e.g.:

```C
/* co_tmr.h */
#ifndef CO_TMR_TICKS_PER_SEC
#define CO_TMR_TICKS_PER_SEC     100
#endif
```

Optionally, a `static_assert` could be used to guarantee that `1000 / CO_TMR_TICKS_PER_SEC` evenly:
```C
_Static_assert(1000 % CO_TMR_TICKS_PER_SEC == 0, "Bad configuration for CO_TMR_TICKS_PER_SEC");
```
Note that `_Static_assert` is part of the C11 standard.
I do not know which range of compilers / language features you wish to support in this project, but that might be something to take into consideration.

Finally, this change could introduce potential issues inside the definition inside this macro:
```C
#define CO_TMR_TICKS(ms)         ((ms)/(1000/CO_TMR_TICKS_PER_SEC))
```
If `CO_TMR_TICKS_PER_SEC` is set to 10000 in order to allow a time granularity of 100us, the expression `1000/CO_TMR_TICKS_PER_SEC` would evaluate to 0, leading to a divide-by-zero bug. FreeRTOS has a macro called `pdMS_TO_TICKS()` with similar issues; [this thread](https://www.freertos.org/FreeRTOS_Support_Forum_Archive/February_2016/freertos_On_pdMS_TO_TICKS_macro_definition_317c7160j.html) discusses some of the design implications. In the case of this project, it might be useful to add a second macro `CO_TMR_US_TO_TICKS` for situations where fine-grained timing is possible.

michael-hillmann, 2020-09-26:
The longterm goal is to get a library, which depends on the microcontroller, not on the specific project needs. Therefore we should try to remove all compiler preprocessor configuration stuff. You are right - this configuration `CO_TMR_TICKS_PER_SEC` should be changable for specific project needs.

Your proposal is a good fast workaround. For the final solution I prefer a way which will not limit the library to a single value of `CO_TMR_TICKS_PER_SEC`. Some thoughts:
1. the value of `CO_TMR_TICKS_PER_SEC` must describe the real tick rate of the hardware timer
2. as user I would be happy if there is no need to set defines as "description of the hardware" and the stack detect the value

What would you think on a design change with the goal to define the hardware timer connection with a device driver interface. This device driver can handle the setup of the hardware timer and can return the current setup easily.

cesarvandevelde, 2020-09-27:
I agree with you, it would be good to move towards a driver-style hardware 
interface. Personally, I like the way [littlefs](https://github.com/ARMmbed/littlefs) handles hardware-specifics in their library. In short, application-specific functions and constants are contained within a single struct, and this struct is passed to the library on initialization. I have not used littlefs yet (planning though!), so I cannot speak to the specifics.

In the context of canopen-stack, there are two main use cases to consider:
* Bare-metal: everything runs inside a superloop and a hardware timer is used to kick off time-triggered CAN events.
* RTOS: in this case, the system already provides a bunch of timing primitives, so the time basis for canopen-stack is not necessarily a "raw" hardware timer.

In my application, I am working with FreeRTOS with the tick rate set to 1000Hz. I am using `vApplicationTickHook()` instead of a dedicated timer, as this is essentially a free timer. I am running the entire canopen-stack inside a single task. This task blocks on two event flags: message received and timer expired. These flags are set from ISRs.

```C++
extern "C" void vApplicationTickHook() {
    static uint8_t counter = 0;
    BaseType_t higher_priority_task_woken = pdFALSE;

    if (counter < 9) {
        counter++;
    } else {
        counter = 0;

        if (COTmrService(&hmi_input_node.Tmr) > 0) {
            can_task.notify_from_isr(
                CanTaskEvent::timer_elapsed,
                &higher_priority_task_woken
            );
        }
    }

    portYIELD_FROM_ISR(higher_priority_task_woken);
}
```

In my usage case (i.e. with RTOS), there were two things that I found annoying:
1) The FreeRTOS tick rate (1000Hz) is different from the canopen tick rate (100Hz), so I had to add a divide-by-10 step in software. Not a big deal, but it feels like wasted potential. The system can provide `1ms` time granularity, but only `10ms` granularity is used. 
2) The callbacks `COTmrLock()` / `COTmrUnlock()` are called by `COTmrService()` inside ISR-context. In my system, `COTmrService()` cannot be interrupted by another `COTmr` function, and I do not want to disable the RTOS tick interrupt from ISR. So my `COTmrLock()` / `COTmrUnlock()` implementation checks the execution context, and only enters a critical section if it is called in thread mode.

On the other hand, I really like that the return value of `COTmrService()` indicates whether `COTmrProcess()` is required. In general, integration was very straigtforward!

For reference, the approach of CanFestival is much more oriented at hardware timers. They use a separate config file for timer-related constants:

```C
#define TIMEVAL uint32_t

// using 16 bits timer
#define TIMEVAL_MAX 0xFFFF

// The timer is incrementing every 10 us.
#define MS_TO_TIMEVAL(ms) ((ms) * 100)
#define US_TO_TIMEVAL(us) ((us) / 10)
```

Integration is handled through callback functions `setTimer()`, `getElapsedTime()`, and `TimeDispatch()`. This stack also uses an internal software timer implementation to create multiple concurrent timers.

As a second point of reference, the Lely CANopen stack uses `timeval` structs from `sys/time.h` extensively. See [here](https://gitlab.com/lely_industries/lpc17xx) for an example of how their system works.

michael-hillmann, 2020-10-01:
At a first step (for v4.0.3), I implement your proposal:

```c
/* co_tmr.h */
#ifndef CO_TMR_TICKS_PER_SEC
#define CO_TMR_TICKS_PER_SEC     100
#endif
```

with simple handling for values greater than 1000. The next feature release (v4.1.0) will include a more flexible driver interface.

closed 2020-10-10 by michael-hillmann