embedded-office/canopen-stack #14: Compile-time warnings: declared inside parameter list, incompatible-pointer-types, unused-but-set-variable

cesarvandevelde, 2020-09-24:
**Description**

The compiler emits 3 distinct types of warnings when compiling the stack: 

1. **declared inside parameter list**: 
    ```
    In file included from ../dependencies/canopen-stack/canopen/include/co_nmt.h:28,
                     from ../dependencies/canopen-stack/canopen/source/co_nmt.c:21:
    ../dependencies/canopen-stack/canopen/include/co_obj.h:276:56: warning: 'struct CO_NODE_T' declared inside parameter list will not be visible outside of this definition or declaration
      276 | typedef uint32_t (*CO_OBJ_SIZE_FUNC) (CO_OBJ *, struct CO_NODE_T *, uint32_t);
          |                                                        ^~~~~~~~~
    ```
2. **-Wincompatible-pointer-types**:
    ```
    ../dependencies/canopen-stack/canopen/source/co_nmt.c:72:42: warning: initialization of 'int16_t (*)(CO_OBJ *, struct CO_NODE_T *, void *, uint32_t)' {aka 'short int (*)(struct CO_OBJ_T *, struct CO_NODE_T *, void *, long unsigned int)'} from incompatible pointer type 'int16_t (*)(CO_OBJ *, struct CO_NODE_T *, void *, uint32_t)' {aka 'short int (*)(struct CO_OBJ_T *, struct CO_NODE_T *, void *, long unsigned int)'} [-Wincompatible-pointer-types]
       72 | const CO_OBJ_TYPE COTNmtHbCons = { 0, 0, COTypeNmtHbConsRead, COTypeNmtHbConsWrite };
          |                                          ^~~~~~~~~~~~~~~~~~~
    ```
3. **-Wunused-but-set-variable**:
    ```
    ../dependencies/canopen-stack/canopen/source/co_pdo.c:908:16: warning: variable 'cod' set but not used [-Wunused-but-set-variable]
      908 |     CO_DICT   *cod;
          |                ^~~
    ```

As I understand it, issues 1 and 2 are caused by the order of definitions: the function prototypes that take CO structs as arguments are declared before the definition of those structs. To solve these issues, the header files will need to be restructured.

Issue 3 is much simpler to solve: variable `cod` should be removed from `COTypePdoComIdWrite` and inside `COTypeNmtHbConsWrite`.

**Steps to Reproduce**

Below is a verbose build log. The log file includes the GCC invocations used to compile the CANopen stack:

[build.log](https://github.com/embedded-office/canopen-stack/files/5275076/build.log)

**Expected behavior**

The stack should compile without warnings. Compiler warnings usually point to underlying issues that can unexpectedly become real problems.

**Environment**

- Target: STM32F103
- Cross-Compiler: GNU Arm embedded toolchain 9-2020-q2-update
- CANopen Stack: 4.0.2

shersey-locus, 2020-09-25:
I'm seeing similar warnings/errors using Keil uVision 5 targeted for an STM32F405, but I'm just getting started setting things up, so it could be a PBKAC on my part. Relevant messages are:

```
Build started: Project: AFBR_S50_TOF_rangefinder
*** Using Compiler 'V5.06 update 6 (build 750)', folder: 'C:\Keil_v5\ARM\ARMCC\Bin'
...
compiling co_nmt.c...
Drivers/canopen/include/co_obj.h(276): warning:  #231-D: declaration is not visible outside of function
  typedef uint32_t (*CO_OBJ_SIZE_FUNC) (CO_OBJ *, struct CO_NODE_T *, uint32_t);
...
Drivers\canopen\source\co_nmt.c(72): warning:  #144-D: a value of type "int16_t (*)(CO_OBJ *, struct CO_NODE_T *, void *, uint32_t)" cannot be used to initialize an entity of type "CO_OBJ_READ_FUNC"
...
Drivers\canopen\source\co_nmt.c(299): error:  #167: argument of type "CO_NODE *" is incompatible with parameter of type "struct CO_NODE_T *"
      err = COObjRdValue(obj, node, &num, CO_BYTE, 0);
Drivers\canopen\source\co_nmt.c(470): warning:  #550-D: variable "cod"  was set but never used
      CO_DICT     *cod;

```

shersey-locus, 2020-09-25:
Adding a forward struct declaration "struct CO_NODE_T;" to line 251 of "co_obj.h" resolved the above-mentioned warnings, and brought me to a lot of undefined-symbol linker errors, which is what I would expect as I have not yet coded the system-specific interface functions.

michael-hillmann, 2020-09-26:
Ok, I currently use compiler `Microsoft Visual C++ V19.24` for my development and test environment. Unfortunately, I'm new in using "Modern Cmake" and I didn't realize that I need to enable the warnings explicitly. In the branch "develop" I fix all warnings, which rises up after enabling all warnings in MSVC.

Could you both check if this covers all warnings in your environments as well, please?

cesarvandevelde, 2020-09-27:
The develop branch fixes most warnings in GCC. Here's the build output:

```
[23/90] Building C object dependencies/CMakeFiles/canopen.dir/canopen-stack/canopen/source/co_nmt.c.o
../dependencies/canopen-stack/canopen/source/co_nmt.c: In function 'COTypeNmtHbConsWrite':
../dependencies/canopen-stack/canopen/source/co_nmt.c:470:18: warning: variable 'cod' set but not used [-Wunused-but-set-variable]
  470 |     CO_DICT     *cod;
      |                  ^~~
[37/90] Building C object dependencies/CMakeFiles/canopen.dir/canopen-stack/canopen/source/co_pdo.c.o
../dependencies/canopen-stack/canopen/source/co_pdo.c: In function 'COTypePdoComIdWrite':
../dependencies/canopen-stack/canopen/source/co_pdo.c:908:16: warning: variable 'cod' set but not used [-Wunused-but-set-variable]
  908 |     CO_DICT   *cod;
      |                ^~~
```

These last two warnings are easy to fix: the object dictionary pointer is never used, and can be deleted without consequence.

```diff
diff --git a/canopen/source/co_nmt.c b/canopen/source/co_nmt.c
index b9787e2..669b6e8 100644
--- a/canopen/source/co_nmt.c
+++ b/canopen/source/co_nmt.c
@@ -467,14 +467,12 @@ void CONmtHbConsMonitor(void *parg)
 */
 int16_t COTypeNmtHbConsWrite(CO_OBJ* obj, struct CO_NODE_T *node, void *buf, uint32_t size)
 {
-    CO_DICT     *cod;
     CO_HBCONS  *hbc;
     int16_t     result = CO_ERR_TYPE_WR;
     uint32_t    value  = 0;
     uint16_t    time;
     uint8_t     nodeid;
 
-    cod  = &node->Dict;
     hbc  = (CO_HBCONS *)obj->Data;
     if (hbc == 0) {
         node->Error = CO_ERR_CFG_1016;
diff --git a/canopen/source/co_pdo.c b/canopen/source/co_pdo.c
index 5d4b325..a148728 100644
--- a/canopen/source/co_pdo.c
+++ b/canopen/source/co_pdo.c
@@ -905,7 +905,6 @@ int16_t COTypePdoMapWrite(CO_OBJ *obj, struct CO_NODE_T *node, void *buf, uint32
 */
 int16_t COTypePdoComIdWrite(CO_OBJ* obj, struct CO_NODE_T *node, void *buf, uint32_t size)
 {
-    CO_DICT   *cod;
     CO_NMT   *nmt;
     CO_TPDO  *tpdo = 0;
     CO_RPDO  *rpdo = 0;
@@ -928,7 +927,6 @@ int16_t COTypePdoComIdWrite(CO_OBJ* obj, struct CO_NODE_T *node, void *buf, uint
         return (CO_ERR_OBJ_RANGE);
     }
 
-    cod     = &node->Dict;
     nmt     = &node->Nmt;
     pcomidx = CO_GET_IDX(obj->Key);
     if ((pcomidx >= 0x1400) && (pcomidx <= 0x15FF)) {
```

michael-hillmann, 2020-09-28:
Hmm, with all warnings enabled (`/W4 /WX`) in MSVC v19.23.28106.4 for x64, the warning [C4189](https://docs.microsoft.com/en-us/cpp/error-messages/compiler-warnings/compiler-warning-level-4-c4189?view=vs-2019) is not detected; Strange!?

I think it is best to set up a parallel GCC environment for checking the warning-free compilation with future changes.

shersey-locus, 2020-09-28:
I also get only those two warnings for the unused object dictionary pointer "cod" under Keil uVision with the updated code. Looks like you're on the right track. Thanks for maintaining this package!

michael-hillmann, 2020-09-29:
Ok, I setup my local environment to IAR EWARM and see the same remaining warnings. Now the develop branch is warning-free.

cesarvandevelde, 2020-09-29:
Great, everything compiles without warnings on GCC as well!

A CI job could be used to compile/check GCC output/test automatically.
I've set this up in the past with Gitlab CI, though I'm not really with the way it works in Gitlab Actions...

michael-hillmann, 2020-10-01:
fixed in v4.0.3

closed 2020-10-01 by michael-hillmann