embedded-office/canopen-stack #15: Requesting a fully-populated source example

shersey-locus, 2020-09-30:
I'm an experienced embedded developer, but entirely new to using CAN and CANopen, and I'm trying to use this CANopen stack to implement my first CANopen-based device. It's turning out to be a steep learning curve.
Because I'm new to CANopen, there are many assumptions baked into the various support stacks that I haven't yet learned (for example, are all of the services supported by the stack required, or are some optional? Can I build an application without a heartbeat or LSS? How to determine how many timers, SDO buffers are needed? How do I correctly report an error with the emergency table, respond to incoming RPDOs, trigger outgoing TPDOs, safely update variables that the object dictionary refers to?) 

One of the resources I generally look for is a fully-worked-out sample application (such as the clock application listed in the documents for this stack). A complete set of source code, even for an MCU architecture different from those I use, always provides insight into how to successfully use the software stack in question, and generally reveals things I didn't learn from the documentation.

Unfortunately, I'm unable to find any complete source code examples whatever for this CANopen stack. There are various sets of documentation on the 'Net that relate to this stack, but they vary in quality and in how well they actually match the API in this repository, and none of them provide the integrated snapshot that a simple full source example delivers.

I'm acutely aware that questions always outnumber answers, and I always hesitate to ask others to do unpaid work on my behalf, so I shall limit myself to suggesting that including a full source example such as the clock would be of great help to those new to the stack.

michael-hillmann, 2020-09-30:
Thank you for your input. As I'm working for many years on CANopen topics, the documentation may assume too much knowledge from the CANopen specification(s). This is a good point for improvement because I would like to see, that this CANopen library is helping embedded developers to start quite smoothly into the world of CANopen.

So I would try to include an example in full source code (as I already have this running on my PC), but for simplicity with a simulated CAN interface (already set up and used in the test suite). This could provide a nice starting point for investigation of application development with a CANopen stack.

The step to the microcontroller is very specific on the CAN modules and existing device drivers. I must think about that - maybe I get an idea or some help from someone with a real microcontroller on the desk. I have only some projects with customer hardware, therefore I'm not allowed to publish the software to the public.

Stay tuned, the example is available soon.

shersey-locus, 2020-09-30:
Thanks. It's almost a Zen koan that the person who knows most about a project has the hardest time writing for an audience who lack the background knowledge the expert no longer notices is there. 

From a user's perspective, the system-specific interface is the least troublesome part. As an implementer, I can reasonably be expected to know or discover the details of operating the CAN hardware for my chosen platform, so you've already done your job there by pointing me to the hooks required to talk to it. (Although a bit of explanation about the other system-specific helper functions such as COIfReceive, COLssLoad, COParaLoad, CONmtHbConsChange and the like would also be useful, so a user knows which ones must be populated and why, and which can be left as empty shells).

A source example that is complete except for those system support routines would provide excellent instructional value, so I would think you could safely stub those out as /* FILL IN YOUR STUFF HERE */, and the example would be just as useful.

The test suite is of some help to me, but its usefulness is limited since the test functions basically verify the API rather than showing how a real-world application would use them.

Thanks for the support of this project in general, and this question in particular!

michael-hillmann, 2020-10-08:
If you may have some time, you can have a look into the branch `develop`. There is some enhancements on driver side (missing more documentation) - but the example and callbacks are now documented as you suggested.

Any feedback is welcome!

shersey-locus, 2020-10-08:
This is EXCELLENT. I'm elbow-deep in some other code this week, so it will be several days before I can resume work on my CANopen project (isn't that just the way it always goes ;-), but my quick look at the example code and API docs has already shown me things I need to do differently. This additional information is going to make a major difference and save me days of flailing about. Thank you VERY much.

shersey-locus, 2020-10-08:
One lingering question (which may be answered in documentation that I haven't yet read): I'm not sure how to properly use the APIs to access a remote node's dictionary objects as PDOs and SDOs. In the clock example, I would expect that either the clock would asynchronously report time to the world at large using a TPDO, or that a remote node would query it using either PDO or SDO constructs. My end application involves remotely setting device parameters with the SDO mechanism, and sending cyclic measurement data using the PDO mechanism.

michael-hillmann, 2020-10-12:
When your application must set device parameters on remote node dictionaries, you typically need an SDO Client (which is not supported by the Canopen-Stack Project). For single expedited transfers, you may emulate the requests with `COIfCanSend()` to set data in the remote dictionary:

```c
/* write a small function for encoding: */
void MySdoRequest(CO_IF *cif, uint8_t nodeId, uint16_t index, uint8_t subindex, uint32_t data)
{
    CO_IF_FRM frm;

    frm.Identifier = 0x600u + nodeId;  /* COBID for SDO server on given Node */
    frm.DLC = 8u;
    frm.Data[0u] = (uint8_t)0x2Fu;    /* writing a byte - use 0x2Bu for a word and 0x23u for a long */
    frm.Data[1u] = (uint8_t)(index);
    frm.Data[2u] = (uint8_t)(index >> 8u);
    frm.Data[3u] = (uint8_t)(subindex);
    frm.Data[4u] = (uint8_t)(data);
    frm.Data[5u] = (uint8_t)(data >> 8);
    frm.Data[6u] = (uint8_t)(data >> 16);
    frm.Data[7u] = (uint8_t)(data >> 24);
    bytes = COIfCanSend(cif, &frm);
    if (bytes != sizeof(CO_IF_FRM)) {
        /* cannot send SDO request */
    }
}
```

You can trigger the PDO measurement transmission with one of the API functions, described in [TPDO Event](https://canopen-stack.org/docs/api/tpdo).

michael-hillmann, 2020-10-12:
Example and documentation improved. Feel free to open new issues for further questions.

closed 2020-10-12 by michael-hillmann