embedded-office/canopen-stack #18: Design choices question (driver interface)

moe123, 2020-10-11:
Hello, why don't you make a timer, mem, bus interfaces based on callbacks? it would let rest of the code build and run perfectly; there are just simple vtables (struct) handling n-delegated-operations?

michael-hillmann, 2020-10-12:
Cool, that you are looking and thinking about the design. Unfortunately, I don't understand exactly what you mean by "simple vtables (struct) handling n-delegated-operations"; could you give me an example?

The reason for the current approach is the support for multiple communication interfaces (like we need for gateways) and the consistent handling of all drivers. If you see an easier path to get these needs managed - I'm open to thinking about optimizing the current implementation.

moe123, 2020-10-12:
@hello something like this

```
typedef void     (*canopen_timer_start)  int id);
typedef uint8_t  (*anopen_timer_update) (int id);
typedef uint32_t (*anopen_timer_delay)  (int id);
typedef          (*anopen_timer_reload) (int id, uint32_t reload);

#define canopen_timer_decl(ns) typedef struct canopen_##ns##_timer \
{
	canopen_timer_invalidate u_invalidate; \
	canopen_timer_delay      u_delay; \
	canopen_timer_stop       u_stop; \
	canopen_timer_start      u_start; \
	canopen_timer_update     u_update; \
}

#define canopen_timer_init(timer, id, freq \
	invalidate \
	delay \
	stop \
	start \
	update \
) \
	timer->u_invalidate = invalidate;
	...

canopen_timer_decl(something) timer;
canopen_timer_init(&timer, something_id, ...);

then you can have a queue of driver ids.
```

michael-hillmann, 2020-10-12:
Let me explain, how I understand your suggestion (just to be sure, that we are talking about the same stuff):

We extend the declaration of driver interface function types from the [current implementation](https://github.com/embedded-office/canopen-stack/blob/master/canopen/include/co_if_timer.h#L36) with an additional argument `int id`:

```c
typedef void     (*CO_IF_TIMER_INIT_FUNC)   (int id, uint32_t freq);
typedef void     (*CO_IF_TIMER_RELOAD_FUNC) (int id, uint32_t reload);
typedef uint32_t (*CO_IF_TIMER_DELAY_FUNC)  (int id);
typedef void     (*CO_IF_TIMER_STOP_FUNC)   (int id);
typedef void     (*CO_IF_TIMER_START_FUNC)  (int id);
typedef uint8_t  (*CO_IF_TIMER_UPDATE_FUNC) (int id);
```

Instead of declaring a single vtable structure for the used timer, we declare a macro to allow each driver allocating its own global vtable structure (which identifies itself by the given `device_name`):

```c
#define CO_IF_TIMER_DRV_DECL(device_name)          \
  typedef struct CO_##device_name##_TIMER_DRV_T {  \
    CO_IF_TIMER_INIT_FUNC   Init;                  \
    CO_IF_TIMER_RELOAD_FUNC Reload;                \
    CO_IF_TIMER_DELAY_FUNC  Delay;                 \
    CO_IF_TIMER_STOP_FUNC   Stop;                  \
    CO_IF_TIMER_START_FUNC  Start;                 \
    CO_IF_TIMER_UPDATE_FUNC Update;                \
  } 
```

Now, I become unsure: we provide another macro, which allows us to initialize the global variable (what shall we do with the argument `id`?)

```c
#define CO_IF_TIMER_DRV_SET(timer, id, init, reload, delay, stop, start, update) \
  do {                      \
    timer->Init = init;     \
    timer->Reload = reload; \
    timer->Delay = delay;   \
    timer->Stop = stop;     \
    timer->Start = start;   \
    timer->Update = update; \
  } while(0)
```

Within a specific driver implementation (let's say for 'stm32f429', we allocate a global variable (which in fact holds the vtable structure) and initializes the function pointers with the implemented `DrvTimerX()` functions:

```c
CO_IF_TIMER_DRV_DECL(stm32f429);
CO_IF_TIMER_DRV_SET(&stm32f429, id,
   DrvTimerInit,
   DrvTimerReload,
   DrvTimerDelay,
   DrvTimerStop,
   DrvTimerStart,
   DrvTimerUpdate);
```

Hmm, I guess I miss something. What should I do with the `id`? How can the common code use the driver (as the common code didn't know the specific name of a driver - here 'stm32f429').

moe123, 2020-10-12:
simple stm32f429 is an integer id mapped to name_space; it's all about multiplexing; i.e many-to-one-to-many which is what is the main role of a driver; the rest of of code should only handle and care about the protocol stricto-censo i.e "msg routing"; this is really not the job of a message protocol to test at runtime if its environment  has been well setup; should be already in a trusted relationship when it happens.

michael-hillmann, 2020-10-12:
Hmm, you are right - currently there is no check of the driver function pointers before using them. Do you think we should check the driver pointers during initialization? This would be a performance overhead, which we should avoid. Especially the small 8bit microcontrollers will stop working with sufficient performance.
Maybe we should continue with a more simple way: we can remove the `co_if_xx` files and declare the implemented functions `COIfxx()` as static callbacks, which must be provided as external symbol. Well, this would introduce some logic into the driver when building for multiple communication lines - ergo: not so good ;-(

moe123, 2020-10-12:
@hello, I don't pretend to hold the grand truth; however by emplacing back each responsibility to the right owner; I thing it's a good start; I am not worried about 8-bit controllers/2kB in that matter; "we should check the driver pointers during initialization?" yes If "debug" if you wish to; else, it's unlikely to ever happen "here the ruleset: comply or or die"; + multiple communication lines can be handled at build time.

michael-hillmann, 2020-10-12:
Thank you for your input.

closed 2020-10-12 by michael-hillmann