embedded-office/canopen-stack #19: Recommended clarification / improvement to quickstart example, and some suggested macro-foo

shersey-locus, 2020-10-22:
I've now managed to bring up a working implementation of this CANopen stack on my STM32F405 development hardware. Thank you very much for this software stack, and for doing additional work on the documentation in response to user requests. It is greatly appreciated and extremely helpful.

I've used the quickstart example as a pattern for development, and I have several observations and suggestions from the perspective of a novice user.

I had significant trouble properly setting up TPDOs in the object dictionary; the example contains an error there, and there are also opportunities to make this setup easier. (Some of my difficulties came from not really understanding how the PDO configuration mechanisms are intended to work, of course. The only real cure for that was reading and research.)

The example object dictionary contains the following TPDO communications entries for TPDO #0:
```
    {CO_KEY(0x1800, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)2},
    {CO_KEY(0x1800, 1, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)0x40000180},
    {CO_KEY(0x1800, 2, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)254},
```

The entry at subindex 1 encodes the COBID 0x180 that is used for TPDO 0 frames, but there are problems with it:
- The value here is plugged in as a "magic number" that is quite confusing to the newcomer; it could easily be generated by a macro instead (see below), reducing the possibility of user error and improving readability
- The COBID 0x180 needs to be indexed by the node ID to get the correct COBID value for a real node; I think that ``CO_OBJ_D__R_`` should be replaced by ``CO_OBJ_DN_R_`` to accomplish this. (In my application, node IDs are generated at run time by hardware straps, so hard-coding the node ID into the magic number is not possible.)

I wrote these macros to simplify the process:
```
// PDO communication entries encode the node number and a service flag with the basic TPDO COBID.
// These macros generate the right COBID for TPDO x or RPDO x, and the dictionary does the node-ID indexing.
#define PDO_NON_RTR_FLAG 0x40000000
#define PDO_COBID_INCREMENT 0x100
#define TPDO_COBID_BASE 0x180
#define RPDO_COBID_BASE 0x200
#define TPDO_COBID(x) (uintptr_t)(PDO_NON_RTR_FLAG | (TPDO_COBID_BASE + (x * PDO_COBID_INCREMENT)))
#define RPDO_COBID(x) (uintptr_t)(PDO_NON_RTR_FLAG | (RPDO_COBID_BASE + (x * PDO_COBID_INCREMENT)))
```
Using them, the TPDO comms entries now look like this:
```
    // TPDO #0 communications entries
    {CO_KEY(0x1800, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)2},
    {CO_KEY(0x1800, 1, CO_UNSIGNED32|CO_OBJ_DN_R_), 0, TPDO_COBID(0)},
    {CO_KEY(0x1800, 2, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)254},

```
The example object dictionary contains the following TPDO mapping entries for TPDO #0:
```
    {CO_KEY(0x1A00, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)3},
    {CO_KEY(0x1A00, 1, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)0x21000120},
    {CO_KEY(0x1A00, 2, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)0x21000208},
    {CO_KEY(0x1A00, 3, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)0x21000308},
```
Sub-indices 1..3 simply encode the index/subindex and size of the data that this TPDO will ship out; here, too, these mysterious magic numbers can be made easier to understand with some simple macro magic:
```
// PDO mapping entries have a data field composed of index, subindex, and type
// This handy macro creates that 32-bit entity for us
#define PDO_MAP_ENTRY(x, y, z) (uintptr_t)((x << 16) | (y << 8) | (z))
```
Which gives us the following:
```
    // TPDO #0 mapping entries
    {CO_KEY(0x1A00, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)3},
    {CO_KEY(0x1A00, 1, CO_UNSIGNED32|CO_OBJ_D__R_), 0, PDO_MAP_ENTRY(0x2100, 1, CO_UNSIGNED32},
    {CO_KEY(0x1A00, 2, CO_UNSIGNED32|CO_OBJ_D__R_), 0, PDO_MAP_ENTRY(0x2100, 2, CO_UNSIGNED8}},
    {CO_KEY(0x1A00, 3, CO_UNSIGNED32|CO_OBJ_D__R_), 0, PDO_MAP_ENTRY(0x2100, 3, CO_UNSIGNED8}},
```
I believe this is a bit easier to follow, and makes implementing new entries by hand much easier. (This may be an uncommon use case, and most users may use tools or device profiles to autogenerate the object dictionary, but I don't have convenient access to those tools, and my object dictionary is fairly small.)

As a side note, there are a number of #defines in co_obj.h (lines 38-41 and 46-48) that have the "PDO map" flag, but I have been unable to determine how to use them, or indeed, to determine if they matter in this context. (My code appears to work without them, but I may be fooling myself.) 

I'll also sneak a last question in here, as I haven't been able to figure out the answer on my own: I'm also implementing a relatively basic CANopen master node on an STM32F4 processor; is this stack suitable for use in that role? (If so, I will have to figure out how it should handle incoming TPDO frames from the various remote nodes. A problem for tomorrow...)

Once again, thanks for all the fish!

michael-hillmann, 2020-10-23:
Congratulation to your running stack and thank you for your kind comments.

Regarding your improvement requests I'm happy to do the improvements in the documents for the area PDO mapping configuration and COB-ID generation macros. It's done in the development branch; I hope to release a minor update soon. You are absolutely right, the definitions for COB-IDs are a bit confusing, because the standard defines several additional bits in dependence which COB-ID we want to define. So a good way are a collection of macros, which supports us when defining the CANopen dictionary manually.

The topic: `CO_OBJ_D__R_`  should be replaced by `CO_OBJ_DN_R_` is a bug and must be corrected.

Your macro `PDO_MAP_ENTRY()` is a good way. Note, that the stack already provides a similar macro `CO_LINK()` for exactly this purpose. The missing description of this macro for configuring the PDO mapping is the real issue. I will cover the missing information about the existance of this macro in the configuration section (in a first step); the long term goal is to describe all helpful macros ;-)

*Note: if you want to stay with your macro, please ensure that the 3rd parameter corresponds to the number of bits. The definitions like `CO_UNSIGNED8` are the bit-encoding of sizes for main usage in the object entry key. You could change your macro to:*

`#define PDO_MAP_ENTRY(x, y, z) (uintptr_t)((x << 16) | (y << 8) | (8 << (z >> 4)))`

The #defines in co_obj.h (lines 38-41 and 46-48), which has the `PDO map` in their comment are not for generating the mapping itself. These defines are used for the object entries, which you want to map into a PDO. You should mark these object entries as "PDO mappable" object entries in the corresponding key definition. For example an entry with `CO_UNSIGNED8|CO_OBJ___PR_` is a readonly PDO mappable object entry. Therefore, this entry is useful for TPDO mappings.

I think the stack covers your needs. To process the incoming TPDOs, you just need to define the corresponding RPDOs in your basic master node.

michael-hillmann, 2020-10-23:
Ok, you can find the improvements on the website and in the CANopen Stack release: v4.1.1

closed 2020-10-23 by michael-hillmann

fenugrec, 2021-01-02:
Sorry for bumping this - just wanted to ask @shersey-locus if you progressed on your stm32 implementation; I'm curious how much flash + RAM you're using.

shersey-locus, 2021-01-04:
Yes, my STM32 implementation is up and running. My numbers are a bit fuzzy on account of having other peripheral devices in the mix, but when I filter those out, the CANopen stack with the features I have enabled appears to use about 24K of code + read-only data, and 8K of RAM.