embedded-office/canopen-stack #25: CO_COBID_SDO_STD macro appears to truncate standard CAN ID incorrectly

shersey-locus, 2020-10-29:
The macro definition for CO_COBID_SDO_STD at line 359 of co_obj.h appears to truncate the CAN ID for an SDO server incorrectly. The macro is defined as: 

```
#define CO_COBID_SDO_STD(valid, dynamic, id)       \
    (((uint32_t)(id) & 0x3ffuL)                  | \
     (((uint32_t)(dynamic) & 0x1u) << 30u)       | \
     ((uint32_t)(1uL - ((valid) & 0x1u)) << 31u))

```
The line ``(((uint32_t)(id) & 0x3ffuL)`` truncates a CAN ID of 0x600 to 0x200, which cannot be correct.

As a result, the macros CO_COBID_SDO_REQUEST() and CO_COBID_SDO_RESPONSE() resolve to incorrect COBIDs of 0x200 and 0x180 (plus node ID) instead of 0x600 and 0x580.

Replacing the mask value of ``0x3fful`` with ``0x7fful`` corrects the problem.

michael-hillmann, 2020-10-29:
Is it possible, that you are not using the current version v4.1.2?
This issue is already fixed in the latest version.

shersey-locus, 2020-10-29:
I'll try that. It is indeed possible...

michael-hillmann, 2020-10-29:
Hmm, please clone the master branch instead. I just realize, that the fix is not included in v4.1.2. I will create a new release v4.1.3 tomorrow. Sorry for the confusion ;-(

shersey-locus, 2020-10-29:
Yes, that now works correctly. Thanks!
There once was a chain of no-frills stores in the US named "Building #19", that sold closeouts from a dingy warehouse, with unpainted wood shelving. Their motto was "Confusion is our most important product." ;-)

closed 2020-10-29 by shersey-locus