embedded-office/canopen-stack #28: Need help to understand correct way to use the stack

amrudesh-santhanam, 2020-12-14:
**Description**

Hi. Firstly, thanks for the nice stack. This is by far the best I found. While I am able to get the stack partially working, I am not sure if what I am doing is right. I am using a linux machine and using the virtual can (socket can) before I move it to a micro-controller. I am able to get the quickstart example to send out the time every one second as can messages but I had to do the following to get there.

**Steps to Reproduce**

1. Implemented the timer (co_timer_dummy.c) using `<time.h>`
2. Implemented the can (co_can_dummy.c) using `<linux/can.h>`, `<sys/socket.h>`, `<poll.h>` (read is non-blocking)
3. Implemented the nvm (co_nvm_dummy.c) using a simple `uint8_t` buffer and `memcpy` methods
4. Added `COTmrService(&Clk.Tmr);` to the `while(1)` loop in clock_app.c
5. Initialized a can frame using `CO_IF_FRM time_data;` inside the `static void AppClock(void *p_arg)` method, populated it and sent using `COIfCanSend(&Clk.If, &time_data);`

**Expected behavior**

I was expecting the node to now send the time over can each second and if I send a NMT command to switch it's state to stopped (2) using `cansend vcan0 000#0201` or pre-operational (128) using `cansend vcan0 000#8001` , I was expecting it to stop. 

**Seen behavior**

The node runs but does not send any messages automatically every 1 second. But if I comment out my method to read can messages in the `static int16_t DrvCanRead (CO_IF_FRM *frm)` method and send any message eg: `using cansend vcan0 000#0101` , it starts sending messages but I cant stop it. Problem with this method is that it works only with `CONmtSetMode(&Clk.Nmt, CO_OPERATIONAL);` present before the loop starts. I would not be able switch the state dynamically. When I do read using `DrvCanRead`, I can see using debug that the state actually changes but in this condition, no messages are sent.

I am unsure of the correct way to get things working. Thanks in advance for your help.

**Environment (If Necessary)**

- Linux (Ubuntu 18.04)
- CANopen Stack

amrudesh-santhanam, 2020-12-15:
I got the quickstart to correctly respond to the NMT commands. Timer starts and stops based on my state change commands. The timer implementation had a units issue.

I am still not sure if the `COIfCanSend(&Clk.If, &time_data);` being there in the `AppClock` method is correct. I would assume that the TPDO should construct this message and send rather than me initiating it.

amrudesh-santhanam, 2020-12-15:
Got PDO also working without explicitly calling `COIfCanSend(&Clk.If, &time_data);`

Had to switch transmission type to 0 using `{CO_KEY(0x1800, 2, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)0},`

Yet to test inhibit time and event time parameters.

michael-hillmann, 2020-12-15:
Hi, thanks for testing this CANopen stack for your use case.

I assume there is a small issue in the object dictionary, becasue the sending of the PDO should be triggered by writing to the object entry "2100h:03h (Second)" (this was my intention, but you are right - there are many ways to achieve the transmission of the PDO).

To get my idea running, we need to add the type "CO_TASYNC" to the object entry. With this type, we instruct the CANopen stack to trigger a transmission on each write access to this object entry:
`{CO_KEY(0x2100, 3, CO_UNSIGNED8 |CO_OBJ___PR_), CO_TASYNC, (uintptr_t)&Obj2100_03_08}`

*The transmission type should be 254 in this case and the direct call to COIfCanSend() is not neccessary.*

closed 2020-12-15 by amrudesh-santhanam

amrudesh-santhanam, 2020-12-15:
Thanks for your input. It works perfectly. Sends out the message every second. 

(Missed seeing your comment before I closed the issue)

amrudesh-santhanam, 2020-12-15:
One more query. Regarding SDO. Please see attached screenshot. Here is what happens.

1. I start the application. It sends 701 [1] 00. (which is correct)
2. I send 000 [2] 01 01 to start the node. (works)
3. I send 601 [4] 40 18 10 00 (to request expedited transfer of index 1018, subindex 00)
4. It responds with 581 [4] 4F 18 10 00 (where 4F says that 1 byte is available)
5. I send 601 [5] 40 18 10 00 00
6. It then responds with 581 [5] 4F 18 10 00 04 (where the last 04 is the data)

Is this the expected behavior? Is a second SDO request necessary to retrieve data even in expedited mode? Also, I notice that unless I send the expected number of additional bytes in my request, I dont get the response (5th byte had to be sent as seen above to get the value 04 in the 5th byte in response)

Note: Object dictionary shown above. Leftmost terminal is quickstart application, central terminal is used for candump and right side terminal for cansend.

[image]

amrudesh-santhanam, 2020-12-15:
Also, in the settings for this repository, you should be able to enable discussions. These conversations are better done there as these are not really issues in the code. Please do check.

michael-hillmann, 2020-12-15:
Hi, I just cross checked the specification. In chapter "7.2.4.3.6 Protocol SDO upload initiate" you find the detailed descriptions. Here for reference:

The SDO request needs to have a DLC of 8 with a predefined meaning of all 8 bytes. Well, as you guess, the last 4 bytes are "reserved" and needs to be 0 (for further use).

Currently, the CANopen stack didn't ensure, that the response holds a DLC of 8 (it re-uses the DLC of the request). For safety, we could set the DLC explicitly to 8. This would ensure, that the response is always reasonable (even in the edge case, that the request is not 100% conform to the standard).

amrudesh-santhanam, 2020-12-15:
Thanks. Did not know that. Will do.

michael-hillmann, 2020-12-15:
It's ok, just ask if unsure. I'm happy to collaborate with you!