embedded-office/canopen-stack #29: ballpark numbers for ROM+RAM requirements

fenugrec, 2021-01-02:
Hi,
this is a promising stack and I am seriously considering it for a small project.

I'm very curious to see some typical figures for ROM + RAM requirements for this stack. Of course I understand these will vary a lot depending on enabled features and OD contents, and also on target architecture, but I think even just having a few data points would be valuable to give a rough estimate of what to expect. I understand @shersey-locus has some code running on a stm32F405 from reading some discussions here a while back? That would be one good example.
Another one that would be nice to list is the smallest known, minimal device implementation - just to set a lower limit to adjust expectations.

One can of course do the exercise of configuring + compiling for their target, but I think some basic data could avoid some time loss and frustration.

fenugrec, 2021-01-02:
Preliminary testing on a stm32f042 (cortex-M0), attempting to port the included "clock" demo in `example/quickstart` : 
- at least 12kB ROM (1.5k for the OD which is copied to RAM)
- at least 3.5k RAM (1.5k for the OD;  CO_NODE (sizeof ~ 900B); SdoSrvMem (the huge buffer for segmented / block SDO) 889 B;

RVBS1, 2024-06-20:
Do you, by any chance, still have the code for your port to the STM32F042?

fenugrec, 2024-06-20:
> Do you, by any chance, still have the code for your port to the STM32F042?

Hm, I just have a stale branch in the abandoned non-public project; glossing over it there really wasn't anything special. Just a skeleton with mostly dummy implementations, and calls to `canworker_process()`, `canworker_init()`  etc. from main code.