embedded-office/canopen-stack #32: Question about conformance of this stack

cjardin112, 2021-02-05:
Hi,

I'm currently using your stack for a project on a STM32F0. Works like a charm.

I would like to know if anyone has tested this stack against the CiA conformance test tool ?

Thanks.

Br;

michael-hillmann, 2021-02-07:
I know one company, who checks their product against the CiA conformance test suite.
Are you going to test your product, too?

cjardin112, 2021-02-08:
Alright. Yes I'm planning to test the product. I'll get you updated as soon as I can test it ! :)

hoantv, 2021-03-30:
> Hi,
> 
> I'm currently using your stack for a project on a STM32F0. Works like a charm.
> 
> I would like to know if anyone has tested this stack against the CiA conformance test tool ?
> 
> Thanks.
> 
> Br;

Do you use it with HAL driver.
Currently i write a small library with hal driver. It works but sometime it disconnect to servo drive :(

cjardin112, 2021-03-30:
Hi @hoantv,

I'm currently using it and haven't noticed any connections issues.

Can you explain your problem here ? Do you have any debugger that you can use for investigations ?

Br;

michael-hillmann, 2021-05-04:
Closed due to inactivity and missing data for further investigations

closed 2021-05-04 by michael-hillmann