embedded-office/canopen-stack #37: NVM porting using existing database

amorniroli, 2021-02-10:
**Is your feature request related to a problem? Please describe.**
Currently, the stacks implements it's own "database" module in order to store objects to NVM. NVM abstraction is possible thanks to NVM port driver. Let's say for example a "database" module already exists with his feature etc etc: it would be possible to implement NVM port driver to read/write from/to this "database"? Right now I think it's not, since NVM port driver is really basic and read/write starting from an offset. It's not possible to redirect these calls to existing database.  

**Describe the solution you'd like**
I'd like to have an alternative NVM driver interface, where user can decide to redirect read/write calls to its own database.

**Describe alternatives you've considered**
The solution I thought CO_IF_NVM_READ_FUNC and CO_IF_NVM_WRITE_FUNC to something like:

`typedef uint32_t (*CO_IF_NVM_READ_FUNC )(uint32_t objectId, uint32_t subIndex, uint8_t* ptrBuffer, uint32_t size);`
`typedef uint32_t (*CO_IF_NVM_WRITE_FUNC)(uint32_t objectId, uint32_t subIndex, uint8_t* ptrBuffer, uint32_t size);`

**Additional context**
/

michael-hillmann, 2021-02-11:
Hi, thank you for your enhancement idea!
Can you explain to me more, what data you intend to store in the external database? I have the impression, you want to replace the object dictionary with an external database. This is really difficult, because the read and write activities triggers at some object entries actions (these are managed with the CO_TYPE functions).

The NVM drivers are responsible for saving the current value of a collection of object entries from the dynamic runtime area (RAM) into a persistent area (FLASH, EEPROM, or anything other). These values are used as start values after a restart of the node. If you consider an external database for this kind of persistent storage we could use the argument `start` in the NVM interface as a selector for the given collection of object entries:

Let's try to brainstorm this kind of use-case:

Assuming you have two parameter groups. One with data for the NMT heartbeat configuration and another for the ADC calibration values:

```c
struct NMT_PARA_T {
  uint16_t Obj_1017_0_Heartbeat;
} NmtParaMem;

struct ADC_PARA_T {
  uint8_t Obj_2100_0_AdcFactor;
  uint8_t Obj_2100_1_AdcOffset;
} AdcParaMem;
```

Then we prepare the configuration for save, restore and reset behavior, factory settings, and the `offset` (typical placement in non-volatile memory - for us we could use this as *the identifier of the collection queries*.

```c
const CO_PARA {
  0L,   /* query-id for this parameter group */
  sizeof(NmtParaMem),
  &NmtParaMem,
   :
} NmtParaObj;

const CO_PARA {
  1L,   /* query-id for this parameter group */
  sizeof(AdcParaMem),
  &AdcParaMem,
   :
} AdcParaObj;
```

Note: these parameter group configurations are placed in the object entries 1010h and 1011h.

Now, instead of placing the `NvmParaMem` and `AdcParaMem` with the NVM drivers somewhere in NVM memory, we could store the contained object entries in an external database. First, we need to manage the queries for each entry in the parameter groups:

```c
/* structures to support the query management: */

typedef struct PARA_QUERY_T {   /* data for a single query */
  uint16_t Index;               /* all data we need to access database */
  uint8_t  Subindex;
  uint8_t  Size;
} PARA_QUERY;

typedef struct PG_QUERY_T {   /* structure for a collection of queries */
  PARA_QUERY *Query;          /* start of para query array */
  uint8_t Length;             /* number of para queries */
} PG_QUERY;

/* definition of queries (I would place them near the parameter group definitions): */

PARA_QUERY NmtQueries[1] = {
  { 0x1017, 0x00, 0x2 } /* heartbeat (word) */
};
PARA_QUERY AdcQueries[2] = {
  { 0x2100, 0x00, 0x1 }, /* adc factor (byte) */
  { 0x2100, 0x01, 0x1 }  /* adc offset (byte) */
};

PG_QUERY PgDb[2] = {
  { NmtQueries, 1 },
  { AdcQueries, 2 }
};
```

With the definitions in place, we can prepare the NVM driver for the external database (here only write as an idea):

```c
uint32_t DrvNvmWrite (uint32_t start, uint8_t *buffer, uint32_t size)
{
  uint8_t  idx;
  uint8_t *value = buffer;
  uint8_t  bytes = 0;

  /* loop through addressed query collection and perform database action */
  for (idx = 0; idx < PgDb[start].Len; idx++) {
    pq = PgDb[start].Queries[idx];

    /* place your database query here, e.g.:  */
    ExtDatabaseQuery (pq->Index, pq->Subindex, value, pq->Length);

    /* navigate to next value in parameter memory */
    value += pq->Length;

    /* count all processed bytes */
    bytes += pq->Length;
  }
  return (bytes);
}
```

**!! Note: this is only typed in this issue - never compiled nor tested. I never did this in a real project. It is just the try to investigate, how we can achieve your use-case !!**

amorniroli, 2021-02-12:
Hi @michael-hillmann,

>  I have the impression, you want to replace the object dictionary with an external database

Exactly!

First of all, thanks for your answer: I've just started to use the stack, so I'm still in "learning" phase. I will answer to your in detail next week!

michael-hillmann, 2021-05-04:
Closed due to inactivity, the question is answered.

closed 2021-05-04 by michael-hillmann