embedded-office/canopen-stack #43: PDO Timer Config when entering the OPERATE state.

f1-stuo, 2021-05-05:
Hi,

    Device: [S32K116]
    OS: [none]
    Version [v4.1.5]

I think there's a possible issue when moving to the OPERATE state. If this is done when already in the operational state a new PDO timer is reconfigured on top of the old one, such that the period of the PDO transmission is then incorrect (doubled, tripled or more). This can be reset by transissioning back to pre-operational, then back to operate, but thought it worth flagging up.

Many thanks, and as I've stated perviously, great work!

All the best,

Stu

michael-hillmann, 2021-05-06:
Can you explain in more detail, what you are doing? Are you sending NMT commands twice, or accessing API functions multiple times? I need to investigate this in my environment to fix the software (or documentation) ;-)

smavon, 2021-05-06:
Hi, I found this issue during some tests.

So to give more details on the system : I got a node configured with a PDO in asynchronous mode, with a 1s event timer, working normally in operate state.
Then I send an NMT command to go to the operate state (which, we agree is useless because it is already in).
I was surprised to see the period of my PDO changing : twice more messages.
After this I repeated the send of NMT operate commands, each one adding one new PDO message periodically  send by the device !

I did not kept the traces, but if you can't reproduce this easily, I can illustrate with some captures.
It's not a critical bug, just some curious behavior...

I noticed also that going to pre-operate works normally and after, a new command to go to operate results in a normal behavior.
From the code I imagine that some timers are created for PDOs on the Operate command, which is done even if the node is already in the Operate mode...

Best regards and thanks for this stack ! :)
Sylvain

michael-hillmann, 2021-05-06:
You are right, thank you for the investigation!

**Analysis**
This is a bug in function [CONmtSetMode()](https://github.com/embedded-office/canopen-stack/blob/590685b7a009b722b0edcc923c3985fe31001e88/canopen/source/co_nmt.c#L123). This function initializes the transmit and receive PDOs each time, this function is called.

The correct behavior is: initialize PDOs when entering the operational mode only.

**Solution**
No workaround is possible.
Change of the function for the next release. For example:

~~~c
void CONmtSetMode(CO_NMT *nmt, CO_MODE mode)
{
    if (nmt == 0) {
        CONodeFatalError();
        return;
    }
    if (nmt->Mode != mode) {
        if (mode == CO_OPERATIONAL) {
            COTPdoInit(nmt->Node->TPdo, nmt->Node);
            CORPdoInit(nmt->Node->RPdo, nmt->Node);
        }
        CONmtModeChange(nmt, mode);
    }
    nmt->Mode    = mode;
    nmt->Allowed = CONmtModeObj[mode];
}
~~~

**Testsuite**
We need an additional test in the test suite:
- changing NMT state from operational to operational

commit f349e0fd (2021-07-21): Fix #43 as proposed in issue
closed 2021-07-21 by michael-hillmann