embedded-office/canopen-stack #48: Is it possible to create Object Dictionary in runtime instead of compile time

CoderOnWinks, 2021-06-16:
Instead of static object dictionary, are there any APIs to create the OD on runtime with some input xdd/xml format files, which i can parse and then call the APIs exposed by stack to create dynamic OD.

michael-hillmann, 2021-07-13:
Yes, this is possible. I prepare a documented example in describing how to achieve this. It is not complicated but needs some helper functions to add and remove object entries into the dictionary array. The most important point is, that these functions must ensure that the dictionary is always sorted.

CoderOnWinks, 2021-07-14:
Thanks Michael for considering this feature. I think this will help to create generic CAN products which can ingest the OD fields as configuration files and create OD in memory. Specifically the gateway sort of products which need to talk to multiple different end devices on CAN bus. Can lead to a device agnostic gateway. When is this 4.1.6 milestone planned?

commit bd2006ea (2021-07-21): Add example: Dynamic Object Dictionary Part I and II (fix #48)
closed 2021-07-21 by michael-hillmann