embedded-office/canopen-stack #49: Enable/Disable features of stack

CoderOnWinks, 2021-06-16:
Hi,

Is it possible to enable/disable some of the features of canopen-stack. I am considering not to use NMT, LSS slaveCan i configure them not to build in stack. Please suggest.

Thanks

michael-hillmann, 2021-07-13:
I recommend building a library out of the CANopen stack files. When using the stack, the linker will ensure that only the required functions are included in your image.

closed 2021-07-22 by michael-hillmann

Domen2242, 2021-08-25:
I would also be interested in such a feature as my application is based only on CiA 301. CiA 305's LSS can't be excluded by the linker because it is used in the CONodeProcess function. 
Maybe something like a "#define USE_LSS 1" could be added to the config, which would enable us to save on program space and execution time when we don't to use a feature.
I have created a pull request with my implementation.