embedded-office/canopen-stack #51: Size Preservation During SDO Download

CDK-Vonkil, 2021-07-12:
With an expedited SDO download, the primary success path from COSdoDownloadExpedited( ) calls COObjWrValue( ), but passes a width of CO_LONG.   

Why is the transfer size not preserved?

https://github.com/embedded-office/canopen-stack/blob/590685b7a009b722b0edcc923c3985fe31001e88/canopen/source/co_sdo_srv.c#L360

A similar question applies to COObjWrValue( ), why are all calls to COObjWrType( ) or COObjWrDirect( ) hard-coded as CO_LONG?

This makes a custom CO_TYPE rather problematic, since 1, 2, or 3-byte transfers from a well-formed expedited SDO download are being misreported to the custom type's Write( ) function as 4-bytes long.

michael-hillmann, 2021-07-13:
Good question; this behavior was introduced to handle the data correctly on DSP microcontrollers. Some (or all?) of the DSP cores doesn't allow byte-oriented pointers. Therefore the CO_LONG data, which was given to the COObjWrValue() contains the whole 4-byte value out of the expedited transfers (which is always 4 bytes - independent of the size indication).

To handle the sizes correctly, the COObjWrDirect() will convert the given 4-byte value out of the expedited transfer to the size of the addressed object dictionary entry. This is possible without side effects because the given size indication and the object size are compared before the write operation takes place.

I guess you want to write a custom type and want to handle different sizes with your write function. In this case, the method I would recommend is:

```c
CO_ERR myCustomWrite (struct CO_OBJ_T *obj, struct CO_NODE_T *node, void *buf, uint32_t size)
{
    uint8_t width = CO_GET_SIZE(obj->Key);

    /* now you can handle your data with:
    *  - size (number of valid buffer data)
    *  - width (number of object entry data)
    *  Note: you can trust, that the indicated transfer size (if given) is equal to the width (checked by the SDO server)
    */
}
```

michael-hillmann, 2021-07-13:
Improve documentation to cover this behavior for users who write custom types.

commit 66852c4a (2021-07-21): clarify the argument len in type functions read and write (fix #51)
closed 2021-07-21 by michael-hillmann