embedded-office/canopen-stack #53: SDO block download issue

Domen2242, 2021-07-15:
I have a SDO server node (based on this stack) and a SDO client node (based on canfestival) .
Attempting to do a SDO block download between the devices fails, because the server keeps replying to every sub-block frame with an identical frame - except for ID.
The client rejects these responses and aborts the transfer

Steps to reproduce the behavior:
1. the client initializes a SDO block download (example size 39 bytes, but shouldn't really matter)
2. the server responds successfully
3. the client sends 6 frames of data, setting the 'c' bit on the last frame
4. the server sends 6 response frames. First 5 frames are the same frames received by the client (except for id), while the last one is the proper sub-block acknowledgment frame
5. the client responds to each frame with a abort request, cancelling the transaction before having a chance to parse the correct acknowledgment frame
6. the server responds to each abort frame with an identical frame (except for id)

Expected behavior:
1. the client initializes a SDO block download - 39 bytes
2. the server responds successfully
3. the client sends 6 frames of data, setting the 'c' bit on the last frame
4. the server responds to the last frame with a sub-block acknowledgment frame
5. the client sends a block download end frame
6. the server responds with a block download end response frame, ending the transaction

Modifying the CONodeProcess process function as follows appears to fix the issue.
Replacing
```
    if ((allowed & CO_SDO_ALLOWED) != 0) {
        srv = COSdoCheck(node->Sdo, &frm);
        if (srv != 0) {
            (void)COSdoResponse(srv);
            (void)COIfCanSend(&node->If, &frm);
            allowed = 0;
        }
    }
```
with
```
    if ((allowed & CO_SDO_ALLOWED) != 0) {
        srv = COSdoCheck(node->Sdo, &frm);
        if (srv != 0) {
            CO_ERR status = COSdoResponse(srv);
            if (status != CO_ERR_SDO_ABORT)
                (void)COIfCanSend(&node->If, &frm);
            allowed = 0;
        }
    }
```
yields the expected result. However I'm not sure if this might have any other negative consequences.

michael-hillmann, 2021-07-15:
Thank you for identifying this issue. Your fix looks good, but I would like to integrate a test case for this scenario into the test suite and check that all other tests are still ok.

michael-hillmann, 2021-07-20:
Unfortunately, there are some other related transfers not working correctly, too. I did some further tests and investigations to fix them all.

commit a53afb11 (2021-07-20): Fix #53 in segmented and block transfers
closed 2021-07-21 by michael-hillmann