embedded-office/canopen-stack #56: Custom SDO abort

julienfle, 2021-07-26:
Hello, 

Thanks for the stack.

Is it possible to create a custom SDO abort, by calling the function COSdoAbort with specific code error? 

I would like to have more granularity on SDO errors,that the error corresponds to the application.

Julien

michael-hillmann, 2021-07-27:
Hmm, could you explain to me an example, where you would like to call the COSdoAbort()?
Maybe this is useful for all other users, too?

julienfle, 2021-07-27:
In our system, we have an SDO representing the "mode" of the system
It is possible to change the mode, but only if a few conditions are met:

- the motor must be stopped
- the sensors must have no active signal
- the internal state machine must be in state IDLE
- etc

in case a CAN node tries to change the mode (= write in the SDO "mode") while one of those conditions is unmet, we'd like to abort the SDO and return a detailed error code to explain why it has been aborted:
- motor not stopped => error code XX
- sensor 1 has an active signal => error code YY
- sensor 2 has an active signal => error code ZZ
- etc

michael-hillmann, 2021-07-28:
Ok, in this case, we need to enhance the possibilities within your type-write function(s). Currently, the abort codes are fixed. It is a good point to allow user abort codes, so I think we should introduce in the next release a small macro `CO_SDO_USER_ABORT_CODE(x)` which allows us to return a user-defined SDO transfer abort code from the write function(s).

closed 2021-07-28 by michael-hillmann

michael-hillmann, 2021-07-28:
I found no clear answer in the CiA standard, how to achieve this goal officially. All SDO abort codes are predefined or reserved - so use this feature with care. IMHO, the standard should allow some manufacturer-specific SDO abort codes, too.

julienfle, 2021-07-28:
Thanks for the feature. Yes it's weird that the CiA standard does not accept some manufacturer-specific SDO abort codes. We will use this function knowingly.

commit 09090ec7 (2021-07-28): Merge pull request #57 from embedded-office/feature/user-abort-codes

Introduce user defined SDO abort codes in type functions (fix #56)