embedded-office/canopen-stack #60: Defining more than 4 TPDOs

shersey-locus, 2021-08-19:
The feature description in the top-level README states that the stack supports an unlimited number of TPDOs and RPDOs; however, the code that defines the CAN-IDs for PDOs increments the ID by 0x100 for each PDO, so if I define more than the default 4 TPDOs in this way, the 5th will collide with the CAN-ID of the default server-to-client SDO. I seem to be failing to understand something here. 

What am I missing here? Can I in fact define additional TPDOs without doing major surgery on the stack, and if so, how should I do that? 

Thanks,
Steve

michael-hillmann, 2021-08-19:
I guess you try to use the TPDO COBID macro for getting the standard connection set `CO_TPDO_DEFAULT(0 .. 3)`?

This is no limitation of the CANopen stack, this is just a common recommended set of COBIDs from the CiA standard.

Background: the CANopen standard defines a so-called "standard connection set", which includes 4 TPDOs. This is ok when we assume a network with 128 nodes. This leads us to the needed range of COBIDs: 4 TPDOs x 128 Nodes = 512 COBIDs used for TPDOs. For smaller networks (for example only 16 nodes with NodeId 1..16), we can place more TPDOs within each node: 512 COBIDs / 16 Nodes = 32 TPDOs per node. The reason is the limited number of COBIDs in a CANopen network.

Proposal:
You can create an application-specific (or better: network-specific) connection set. With the _BASE and _INC values you can tweak the COBIDs for your specific use-case:

```c
#define MY_COBID_TPDO_BASE    (uint32_t)0x180
#define MY_COBID_TPDO_INC     (uint32_t)0x8     /* 8 Nodes with NodeId 1..8 */
#define MY_COBID_TPDO_SET(num)  \
    CO_COBID_TPDO_STD(1u, MY_COBID_TPDO_BASE + (num * MY_COBID_TPDO_INC))
```

Alternatively, you can simply set an individual set of COBIDs directly:

```c
#define MY_TPDO1_COBID()   CO_COBID_TPDO_STD(1u, 0x200)   /* 7 Nodes with NodeId 10h, 20h, 30h, .., 70h */
#define MY_TPDO2_COBID()   CO_COBID_TPDO_STD(1u, 0x201)
#define MY_TPDO3_COBID()   CO_COBID_TPDO_STD(1u, 0x202)
  :
#define MY_TPDO16_COBID()   CO_COBID_TPDO_STD(1u, 0x20F)
```

Any scheme is possible.

**Whichever method you use, to get the CANopen network working, you must ensure that no COBID is used multiple times. So it depends on the number of nodes, possible node-IDs, number of TPDOs, and the COBIDs of all other services.**

shersey-locus, 2021-08-19:
Thanks; that makes a great deal of sense. My network is quite small, so I can easily adjust the TPDO COBIDs to accomplish this.

closed 2021-08-19 by shersey-locus