embedded-office/canopen-stack #61: Callback function for reception of data in SDOs

shersey-locus, 2021-08-19:
My application uses a good half-dozen configurable parameters to control a sensor. I've configured them as SDO variables because  of the limited number of default RPDOs, and because it is desirable to be able to set the parameters individually in pre-operational mode as well as operational mode.

However, there's no efficient way to determine when a new parameter value has been received - there is no callback mechanism for SDO transactions as there is for PDOs. As a result, my main loop has to read every parameter from the object dictionary at every sensor update, which I find computationally inefficient. The lack of a callback for SDO operations also makes it difficult to efficiently implement one-time command or parameter-change events that don't involve NMT mode changes. The limited supply of PDOs restricts my ability to just add more RPDOs to handle action-on-value-change events.

My current method for this is to constantly re-read a command code from the object dictionary and compare it to a previous-command-code variable to detect a change, a process I find inelegant at best.

For these reasons, it would be useful to have a callback mechanism analogous to that used for TPDOs and RPDOs that would allow me to trigger internal operations when an SDO transaction has occurred. Is there a reason that such a mechanism would be a bad idea, or architecturally inconsistent with the CANopen device model?

Thanks,
Steve

Domen2242, 2021-08-19:
I solved a similar problem in my project by using a custom object type. See how domain and string types are implemented in the stack as an example. 
You can then call your callback in the custom write function.

shersey-locus, 2021-08-19:
Thanks. I see I'm going to need to wrap my head around domain and string types. My Bigger Project is a firmware update mechanism that writes data received via CANbus to an external flash chip; SDO block transfers would seem like a good choice for that. I get the impression that I can achieve this by defining a custom object type and populating its CO_OBJ_TYPE with my custom read/write/size/ctrl functions.

closed 2021-09-16 by michael-hillmann