embedded-office/canopen-stack #64: How to set up object dictionary entry for a multibyte data buffer?

shersey-locus, 2021-09-01:
I'm adding a firmware update capability to my CANopen devic, and I have a few conceptual questions that I still haven't puzzled out.

My approach here is to define a 256-byte data buffer object in the device, fill it with data via SDO block writes, and then trigger a flash write of the buffer to the external flash by an SDO write to a separate control variable that has a custom CO_OBJ_TYPE. 

The custom CO_OBJ_TYPE for the control variable seems quite straightforward; I mainly need to implement a CO_OBJ_WRITE_FUNC for that object type that uses the new object value to start the flash write operation.

The part that is giving me trouble here is knowing how to correctly define a block of N bytes in the object dictionary so that I can write to it using SDO block write operations. What is the best way to do this?

Thanks again for this software stack and for the help you've provided in using it.

Steve

michael-hillmann, 2021-09-16:
Hi, thank you for your feedback. I try my best in my spare time to provide some help.

For a firmware update capability, I would suggest defining a Domain over the maximum possible size of the **whole** firmware memory region. When uploading the firmware with a block or segmented transfer, the data buffer is already included in the SDO server (with 127 x 7 bytes). The SDO server organizes the write operations (in fact, the FLASH programming sequences), too.

All you must provide is the implementation of the NVM storage driver for your FLASH device.

I will try to provide a small example for a firmware upload. I think this is a helpful example to improve the understanding of the mechanisms.

shersey-locus, 2021-09-17:
Thank you, Michael. That example would be fantastically useful.

commit 6f7abe54 (2021-10-04): add a firmware image upload example description (fix #64)
closed 2021-10-04 by michael-hillmann