embedded-office/canopen-stack #66: Running on 16bit  architectures

nebulorum, 2021-09-16:
I have been trying to run canopen-stack on AVR 16 Bit boards. After hours trying to understand why SPO were not being set, I found out that the Dictionary assumes `uintptr_t` is 32-bit. I was seeing some warning but did not pay attention.  In this case it looses the bit 30 and the SPO is not configured.

Now if I change the `CO_OBJ_T` to use `uint32` on the Data field, I cant map object to the dictionary. I keep getting ` warning: large integer implicitly truncated to unsigned type [-Woverflow]` and I cast twice then I get `error: initializer element is not constant`.  

I would ask: 
1. It there a way to make this with with 16 bit pointers?
2. If this is not supported, show the build fail with an error?

I have to say I really enjoyed the library and the documentation. A couple of things could be improved but I'm overall very impressed. I know these boards may be on the low end, but I'm doing this on a hobby budget. 

I managed to fit it on the 2K RAM board by cutting SPO buffers down (CO_SDO_BUF_SEG  = 15), but I have yet to test using SPOs and have no reference on how tight this can be. I would love to be able to run this stack on Arduino Nanos and MCP2515 cheap boards. Have to say I'm new to CanOpen, but wanted to not reinvent the wheel.

michael-hillmann, 2021-09-16:
Hi, thank you for your feedback.

You are right regarding the 32-bit integers in the dictionary. For the 16-bit micro-controller, we need to consider some implementation-specific details. 

In general, we handle the data of a single dictionary entry in two different ways:
- direct storage (the struct member `uintptr_t data` is used to **store the data** directly)
- standard storage (the struct member `uintptr_t data` is a pointer that **references the data** variable)

> This is designed to reduce the footprint by moving constant data from RAM into FLASH.

For a 16-bit micro-controller, the type `uintptr_t` may be smaller than 32-bit. Due to this fact, we need to consider the following rule:

> The direct storage method for object entries is possible for data values smaller or equal to pointer width (or better: the width of `uintptr_t`). We need to use the standard storage method with a pointer to a variable for data with a higher width.

nebulorum, 2021-09-16:
I did a quick test in changing the type from `uintptr_t` to a Union type:
```
typedef union
{
    intptr_t address;
    uint32_t direct;
} data_t;
```

This allow me to define the Dictionary but my hasty implementation broke the get and set. I think starting trying to setup a TDD approach to this. I think with some care and proper pointer arithmetic  it should be possible to support UINT32 field and handle smaller controllers. This could be a possible route. 

There is a simple test I was playing with: 

```
typedef union
{
    intptr_t address;
    uint32_t direct;
} data_t;

typedef struct
{
    data_t data;
} record_t;

void test() {
    int indirect = 0xcafe;
    record_t rec[] = {
        {0xff1210},
        {&indirect}};
    printf("Direct: %08x\n", rec[0].data.direct);
    printf("Indirect: %08x\n", *((int *)rec[1].data.address));
}
```
Both prints produce the correct output. This is just a toy example, but it could be made to work. Obviously converting a line like this: 
```
*((uint16_t *)val) = (uint16_t)(*((uint32_t*)(obj->Data))&0xffff);
```
Has to be done with care :) My attempt just compiles does not work correctly. 

I also though of having macros and defines to handle differences between big and little endian and 32 or 16 bit pointers. Maybe this would make the code cleaner and more efficient. 

What I'm trying to do is get the Clock app running on a constrained device and see if it still leave a bit of space for doing something useful. I ran into this issues trying to make a SYNC based PDO and not seeing any CanWrites. The clock is ticking correctly and SYNC are accepted. With a dummy CAN driver and just the bare bones its already 79% RAM and 82% of Flash. Maybe not enough for something useful. But there may be fat to cut out. To properly test such a think I still need to figure out how to drive CanOpen (still learning the ecosystem and tool).

nebulorum, 2021-09-21:
After trying a while on PlatformIO I decided to build the project outside my project. I managed to get CMake to build but could not figure our how to run the test. `ctest .`,  `ctest testsuite` or  `testsuite/CanopenTest` don't produce anything (even with what I hope is a broken test). How should I run the test suit?

nebulorum, 2021-09-23:
I got the test to work on my computer ([here](https://github.com/nebulorum/canopen-stack/tree/build-16bit)). I will try to add some test around the idea. But before I add more effort is this something worthwhile doing?

nebulorum, 2021-09-25:
I created test to try out the idea with some safety. I also found some surprising behaviour here are there, maybe they should go in a new issue.

Regarding this comment:
> The direct storage method for object entries is possible for data values smaller or equal to pointer width (or better: the width of uintptr_t). We need to use the standard storage method with a pointer to a variable for data with a higher width.

I will try this out too. But this really is manifested in the CANID for PDO and SDO. The way the sample code was designed it seems cleaner to have them direct. Forcing indirect for 32 bit data in machines with 16bit pointers seems strange. Also would be nice to have some way to fail build if the programmer does not do that correctly.

nebulorum, 2021-09-25:
Your suggestion of using indirect for the large objects worked perfectly! It is a bit clunky since you have to do it for the TPDO Identifier and for the entire mapping. But after that I got to see the TPDO both by SYNC and update. Have to say the library is very feature full. 

Still think the union type could give a more homogeneous API. 

Working with this made me question if there is a way to make the OD reside mostly in Flash (only having updatable fields in RAM).

michael-hillmann, 2021-09-25:
First, of all: thanks for your investigations and contribution!

You are right; your idea with the union type will solve the trouble you have with the 16-bit architecture. As a fact out of the library history, the data type was uint32_t in an old version of this library (3 to 4 years ago). We came into trouble with this data type when the 64-bit architectures rise up and the pointers are wider than 32-bit.

Unfortunately, we change the type to `uintptr_t` in the thinking to capture finally all cases ;-) I didn't use a 16-bit architecture for many years, because the 32-bit devices are cheaper than 16-bit devices and most compilers can be set in a mode where addressing 4GB is mandatory (e.g. leading to 32-bit pointers). Nevertheless, sorry for the inconvenience!

Well, an own data type `co_data`, which is used throughout the whole API and representing your union type would capture all cases. Let's see how smooth it is working in practice. Currently, I'm working on the next release which is scheduled for Oct. 2021. So we can try to get it done without breaking the API for all other users.

nebulorum, 2021-09-25:
I'll give a try with the union test on my fork. Maybe it is more trouble than it is worth.

 I have been in the fence on the 16-bit thing. I'm trying to do this as a hobby and the CAN and 16-bit AVR are very cheap, that is the main reasoning. But if you build your own hardware and they goal is as $ 200 piece of kit it does not make sense to even use these low power CPU.  I really curious  if there is enough out there to get a simple CanOpen system as a hobbyist, but the learning curve is really high. If this does not work I'll move to a larger MCU, may be possible to do a low cost board (STM32F042 + transceiver is around €6 at Mouser).

michael-hillmann, 2021-09-25:
Ok, great! Don't hesitate to ask if you need some assistance.

nebulorum, 2021-09-26:
Branch still needs some clean up but it passed all test on the canopen-stack and worked on my 16-bit pointer environment with minimal change.

The base change was: 
```c
typedef union CO_DATA_T { 
    uint32_t Direct;
    intptr_t Indirect; 
} CO_DATA;
// These define are to make the definitions nicer.
#define CO_DATA_DIRECT(val) ((CO_DATA){.Direct = (val)})
#define CO_DATA_INDIRECT(ptr) (CO_DATA){.Indirect = (intptr_t)(ptr)}
```

Unfortunately I could not get a cleaner on the OD definitions: 
```c
{CO_KEY(0x1001, 0, CO_UNSIGNED8 |CO_OBJ____R_), 0, CO_DATA_INDIRECT(&Obj1001_00_08)},
```
Direct works fine, but for interrect pointer using  `(intprt_t)` my compiler complains with: 
```
error: initializer element is not computable at load time
```

So this is change is not exactly non breaking. I can put a PR once I cleaned a few rough edges but the [branch](https://github.com/nebulorum/canopen-stack/tree/build-16bit) does have the running code.

nebulorum, 2021-10-07:
I added this PR with my proposed changes. 

https://github.com/embedded-office/canopen-stack/pull/72

I think the test need a little bit of cleanup, but they do cover the most important use cases. I have some ideas, but would like an initial feedback before moving forward.

michael-hillmann, 2022-01-28:
I think the cleanest way of solving this issue is a dedicated type for object entries (e.g. CO_DATA). As this will change a lot in stack and documentation, it needs some time to finish the changes. A small prototype is working, but I want to check backward compatibility to avoid changes for all existing users... so stay tuned ;-)

michael-hillmann, 2022-01-28:
Principle idea:

```c
#if sizeof(uintptr_t) < 4
  typedef CO_DATA uint32_t
#else
  typedef CO_DATA uintptr_t
#endif
```

michael-hillmann, 2022-07-12:
fixed with pr #111