embedded-office/canopen-stack #69: Calling COTmrService from timer ISR is causing page fault

trajamohan, 2021-09-24:
Hi thanks for the wonderful project,

I am trying to use the stack in one of my project.

As per the document had implemented the driver for NVM, CAN Interface and Timer.

Though had confusion about which function to call to update timer then from the clock example added the 

`COTmrService(&Node.Tmr);`

in my timer ISR which is called every 1msec 

But when I enable it, is causing page fault at the below line in co_tmr.c

        /* setup next timer event */
        if (tn != 0) {                       
            COIfTimerReload(cif, tn->Delta);       //line causing page fault
        } else {
            COIfTimerStop(cif);
        }
        result = 1;

Am I doing it right?

Environment

STM32
STM32CubeIDE
HAL
FreeRTOS

trajamohan, 2021-09-24:
After going through the timer driver document multiple times currently modified ISR code to call only when the timer driver start event is fired and stopped when timer driver stop event is fired.

So far no issues because the COTmrService(&Node.Tmr); is not called yet on startup, hope the issues will not rise when the stack is fully functional at runtime.

The problem seems to be linked list has no entry (no timer added yet) seems pointer doesn't have valid entry

michael-hillmann, 2021-10-04:
You have analyzed the behavior correctly.
(see [Timer Driver](https://canopen-stack.org/docs/hardware/timer/#timer-start))

closed 2021-10-04 by michael-hillmann