embedded-office/canopen-stack #74: User Object Implementation

sunischit1, 2021-11-16:
### Question : Not able to retrieve the correct value for user objects.

Description : I have to implement user object 0x1F80 (Setting/Getting NMT state) and have followed the approach given in this documentation (https://canopen-stack.org/docs/usecase/dictionary) but not able to retrieve the correct values. Attaching the link of PCAN view in which I am getting the error ([https://drive.google.com/file/d/1zumtuZ8tVzuNXLxdaKzHgN7x1OlVudn5/view?usp=drivesdk]
 
Support Needed : Any specific way to implement these objects and why I am getting this error.

michael-hillmann, 2021-11-16:
Thanks for using the stack. As far as I see in your image you do:

CAN TX: 40 80 1F 01 00 00 00 00  => send a SDO expedited upload requrest for reading the object entry 1F80h subindex 01h
CAN RX: 80 80 1F 01 20 00 00 80  => receive the SDO abort code `CO_SDO_ERR_TOS`

When searching the occurance of this error code in the stack, I found two relevant places. All other places relates to other SDO request types:
1. in `COSdoGetSize()` : Error occurs, when getting the size of the object fails
2. in `COSdoUploadExpedited()` : Error occurs, when reading the value is not returning `CO_ERR_NONE`

As you write, that you are implementing a user object, we would need to see your implementation for further investigations. 

Some hints where you can look:
1. have a look to your size function, which must return the size in bytes
2. return CO_ERR_NONE from your reading function; just have a look to the following tiny example (written without tests) as a orientation how a function could look like:

~~~c
CO_ERR MyTiny32BitRead (struct CO_OBJ_T *obj, struct CO_NODE_T *node, void *buf, uint32_t len)
{
    CO_ERR result = CO_ERR_NONE;

    // get object entry value
    // TODO: adjust this to your user object entry type
    uint32_t objectValue = (uint32_t)(obj->Data);
    uint8_t num = CO_LONG;

    // copy the value into the given buffer
    uint8_t *src = (uint8_t *)&value;
    uint8_t *dst = (uint8_t *)buf;
    while ((len > 0) && (num > 0)) {
       *dst = *src;
       src++;
       dst++;
       len--;
       num--;
    }

    // return successfull operation (no failure considered in this tiny function)
    return (result);
}
~~~

If this doesn't help, you should provide your type definition and all your type functions - so I can have a look ;-)

sunischit1, 2021-11-17:
Thanks Michael, It works for me.

michael-hillmann, 2021-11-17:
Great. Good to hear ;-)

closed 2021-11-17 by michael-hillmann