embedded-office/canopen-stack #75: Mislink in the documentation

sudoDavi, 2021-11-19:
**Describe the bug**
There's a mis-link in the documentation hosted on https://canopen-stack.org/docs

*You get a 404 error when clicking the below mentioned button*
[image]

**To Reproduce**
If you click the Quickstart link on the topics list on the left, it links you to https://canopen-stack.org/examples/quickstart , but the page for the Quickstart section is located on https://canopen-stack.org/quickstart/
[image]

**Expected behavior**
Open the Quickstart page.
[image]

michael-hillmann, 2021-11-26:
Thank you for this issue.
A fix is prepared in the development branch and is deployed with next release.

michael-hillmann, 2022-03-29:
Is constantly improved by switching to Mkdocs (see pull-request #89)

closed 2022-03-29 by michael-hillmann