embedded-office/canopen-stack #77: Is there any Interface to change NodeId and BaudRate ?

sunischit1, 2021-11-24:
I have to update the NodeId and Baud rate using Manufacture Specific CANOpen Object. What Interface I should use to update this. Will it be possible to do so?

I have also tried LSS Commands which is as follows but not able to get the success. See the command list below from PCAN:

- ID=2021,Type=D,Length=8,Data=0401,IDFormat=hex // Switch state global service
- ID=2021,Type=D,Length=8,Data=117C,IDFormat=hex // Set node id to 0x7C
- ID=2021,Type=D,Length=8,Data=17,IDFormat=hex // Store configuration
but while reading using command I am getting the Node Id which is hardcoded in the FW.
- ID=2021,Type=D,Length=8,Data=5E,IDFormat=hex // Reading Node Id
- ID=2020,Type=D,Length=8,Data=5E7F // Responce I got from my slave device.

I am using Embedded Office CANOpen stack version **v40107.**

sudoDavi, 2021-11-24:
First you'll need a CAN driver which supports changing the baudrate, this you'll need to implement since it is application specific and depends on your microcontroller and CAN controller hardware.

Second, since you're talking about changing the baudrate using a _Manufacturer Specific OD Entry_ I take it you are aware of the LSS protocol and are choosing not to use it, correct? If you are unaware of the LSS protocol, it is a protocol specified by the CANopen specifiers that does exactly what you're talking about, sadly the specification of this protocol is still only available for **CiA** members, as you can see in this link [LSS specification: CiA 305](https://can-cia.org/can-knowledge/canopen/cia305/)

Third, I believe (I haven't tested this, but looking at the docs this seems to be feasible), you can do this by implementing the following routine in your application:

1. Start your CANopen node (you'll need to have it already configured with the baudrate the bus is already working with, otherwise it won't be able to communicate)
2. Register a callback when the specified object is modified by the CANopen network
3. Inside the callback, you'll need to stop your CANopen node and re-initialize it with the new baudrate and node-id

sudoDavi, 2021-11-24:
You can see what the documentation says about the interface to control the CANopen node in this page https://canopen-stack.org/docs/usecase/control

sunischit1, 2021-11-25:
Thanks for the Suggestion, Let me give a try.

closed 2021-11-25 by sunischit1