embedded-office/canopen-stack #80: COTmrInsert function TPDO interval timer

NaserFaryad, 2022-01-28:
Hi
I used CANopen-stack on STM32F407. and everything is good. but I found a bug in the TPDO interval timing. I call the `COTmrService(&Clk.Tmr);` every 1 ms in the timer callback function. and the TPDO works well.  
The defined values are as below:

```
/* Define some default values for our CANopen node: */
#define APP_NODE_ID       1u                  /* CANopen node ID             */
#define APP_BAUDRATE      250000u             /* CAN baudrate                */
#define APP_TMR_N         16u                 /* Number of software timers   */
#define APP_TICKS_PER_SEC 1000u               /* Timer clock frequency in Hz */
#define APP_OBJ_N         128u                /* Object dictionary max size  */
```

the following lines in the object dictionary are the transmit PDO communication parameters:

```
    {CO_KEY(0x1800, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)5},
    {CO_KEY(0x1800, 1, CO_UNSIGNED32|CO_OBJ_DN_R_), 0, CO_COBID_TPDO_DEFAULT(0)},
    {CO_KEY(0x1800, 2, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)0xFE},
    {CO_KEY(0x1800, 3, CO_UNSIGNED16|CO_OBJ_D__RW), 0, (uintptr_t)0},
    {CO_KEY(0x1800, 5, CO_UNSIGNED16|CO_OBJ_D__RW), CO_TEVENT, (uintptr_t)60},
```
but when I change the `1800h:05` to any number multiples of ten (for example 10 or 20), TPDO will not be sent periodically anymore. and I receive a TPDO message every millisecond or 9 or other value. 
[image]

I think problem maybe is inside the `COTmrInsert` function. because when I remove the below section of the function, the problem is solved. but I think this part of the function is used for HeartBeat timing.

```
/* time interval is equal to a used timer */
            if (dTnew == dTx) {
                /* add action to used timer */
                tx->ActionEnd->Next = action;
                tx->ActionEnd       = action;
                tn                  = tx;

            /* time interval is before first timer */
            }
```
could you please tell me how to solve this problem?
Thanks

michael-hillmann, 2022-01-28:
Your use-case is quite common and working for years. So I assume it is stable. There is something different. For sure: you shouldn't remove your marked section. This section is intended for handling two actions which needs to be executed at the same time.

When looking into your log, I would assume that we lead into a situation where two timers for the same event timing are existing (which is, of course, not correct).

First question: how do you change the event time? (I assume via CAN SDO protocoll)

Please step through the function `COTypeEventWrite()` for your event timer write operataion which changes the event time. This function is responsible for clearing the existing timer(s) and setting up the new event timer. Maybe we get some information what happens in this situation.

Is there any error registered in your `node->Error` variable?

NaserFaryad, 2022-01-28:
>  how do you change the event time?
I change it  via CAN SDO protocol and also from the source code
`{CO_KEY(0x1800, 5, CO_UNSIGNED16|CO_OBJ_D__RW), CO_TEVENT, (uintptr_t)10}`

> Is there any error registered in your node->Error variable?
I checked it and saw the below errors.
First, this error occurred inside the `COTmrCreate` function.
_CO_ERR_TMR_NO_ACT | no action available while creating_

and after that, the below error occurred too.
_CO_ERR_TPDO_EVENT | error during event timer creation_

Where do you think the problem could be?

michael-hillmann, 2022-01-28:
Seems that you are running out of software timers. Please try to increase the number of available timers/actions in the line:

```c
#define APP_TMR_N         16u                 /* Number of software timers   */
```

NaserFaryad, 2022-01-28:
Thank you for your support

I change `APP_TMR_N` value from 16 to 128. and the errors didn't occur. but the issue still exists. I can't understand why when I set the `1800h:05` value to 9 it works very well, but when I change it to 10, it didn't work. if I'm running out of software timers, it should not work when the `1800h:05` value is 9.

The output log when` 1800h:05 = 9`

[image]

The output log when 1800h:05 = 10

[image]

The below section is a part of `CO_TMR_TIME *COTmrInsert` function. when I set the `1800h:05 = 9`. the `dTnew` is equal to 9 and its value always is 9 until I change  `1800h:05` value. and the `dTx` value variate from 0 to 1000. but when I change the `1800h:05` both of the `dTnew `and `dTx `equal to 10 and the `dTx `value didn't change. 

```
           /* time interval is equal to a used timer */
            if (dTnew == dTx) {
                /* add action to used timer */
                tx->ActionEnd->Next = action;
                tx->ActionEnd       = action;
                tn                  = tx;

            /* time interval is before first timer */
            } else if (dTnew < dTx) {
                /* fetch a timer */
                tn            = tmr->Free;
                tmr->Free     = tn->Next;
                /* setup timer in front of first timer */
                tn->Delta     = dTnew;
                tn->Action    = action;
                tn->ActionEnd = action;
                tn->Next      = tx;
                tmr->Use      = tn;
                tx->Delta     = dTx - dTnew;
                COIfTimerReload(cif, tn->Delta);
            }
```

I think the issue is related to the multiples of 10. 
What is your opinion on this issue?

michael-hillmann, 2022-01-31:
There is a test (`TS_Tmr_Period100ms`) which ensures, that the handling of times with a multiple of 10 is working as expected. Therefore, I guess we should have a look into your timer driver implementation.

As a first step: what happened, if you start with your timing of 9ms and set the timer afterwards (via SDO) to 1000?

NaserFaryad, 2022-01-31:
> There is a test (TS_Tmr_Period100ms) which ensures, that the handling of times with a multiple of 10 is working as expected. Therefore, I guess we should have a look into your timer driver implementation.

I didn't find any document to implement (TS_Tmr_Period100ms). but my  timer driver implementation is like quick start example:
I defined a `static uint32_t TimerCounter = 0u;` timer counter. and the timer driver functions are as below:
```
static void DrvTimerInit(uint32_t freq)
{
    (void)freq;
		TimerCounter = 0u;
    /* TODO: initialize timer, clear counter and keep timer stopped */
}
```
```
static void DrvTimerReload(uint32_t reload)
{
    /* TODO: reload timer counter value with given reload value */
	TimerCounter = reload;
}
```

in the start function, I start the microcontroller timer interrupt. 

```
static void DrvTimerStart(void)
{
    /* TODO: start hardware timer */
	HAL_TIM_Base_Start_IT(&htim2);
}
```

and in the stop function, I set the `TimerCounter ` to zero and also the microcontroller was stoped.

```
static void DrvTimerStop(void)
{
    /* TODO: stop timer and clear counter value */
	HAL_TIM_Base_Stop_IT(&htim2);
	 TimerCounter = 0u;
}
```

```
static uint8_t DrvTimerUpdate(void)
{

	uint8_t result = 0u;

    if (TimerCounter > 0u) {
        TimerCounter--;
        if (TimerCounter == 0u) {
            result = 1u;
        }
    }

    return (result);
}
```

```
static uint32_t DrvTimerDelay(void)
{
	return (TimerCounter);
}
```

In the timer period elapsed callback  function (`HAL_TIM_PeriodElapsedCallback(TIM_HandleTypeDef* htim)`) that every 1 ms was called, I put the` COTmrService(&Clk.Tmr);` function. it mean the `COTmrService(&Clk.Tmr);` function was called every 1 ms.

```
void HAL_TIM_PeriodElapsedCallback(TIM_HandleTypeDef* htim)
{
    COTmrService(&Clk.Tmr);
	  
}
```

> As a first step: what happened, if you start with your timing of 9ms and set the timer afterwards (via SDO) to 1000?
When I start with  `1800h:05 = 9 ms` and set the  `1800h:05 = 1000 ms`(via SDO) I will have the output as below images.

[image]

michael-hillmann, 2022-01-31:
Your timer driver looks good. If I setup a node with this timer driver (on a STM32F769) it just works. While setting up the quickstart with your environment and just extending the TPDO communication settings, I got similar effects. After some investigation, I identified the problem in my case: I mixed up the TPDO transmission with `CO_TEVENT` **and** `CO_TASYNC`. This is not working. After removing the `CO_TASYNC` from the object entry, the timing is stable and correct.

Please double-check, that you do configure multiple triggers for a single TPDO transmission?

michael-hillmann, 2022-01-31:
To be 100% clear:

**bad config**
```c
/* define the static object dictionary */
static struct CO_OBJ_T ClockOD[APP_OBJ_N] = {
    {CO_KEY(0x1000, 0, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)0},
    {CO_KEY(0x1001, 0, CO_UNSIGNED8 |CO_OBJ____R_), 0, (uintptr_t)&Obj1001_00_08},
    {CO_KEY(0x1005, 0, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)0x80},
    {CO_KEY(0x1017, 0, CO_UNSIGNED16|CO_OBJ_D__R_), 0, (uintptr_t)0},

    {CO_KEY(0x1018, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)4},
    {CO_KEY(0x1018, 1, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)0},
    {CO_KEY(0x1018, 2, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)0},
    {CO_KEY(0x1018, 3, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)0},
    {CO_KEY(0x1018, 4, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)0},

    {CO_KEY(0x1200, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)2},
    {CO_KEY(0x1200, 1, CO_UNSIGNED32|CO_OBJ_DN_R_), 0, CO_COBID_SDO_REQUEST()},
    {CO_KEY(0x1200, 2, CO_UNSIGNED32|CO_OBJ_DN_R_), 0, CO_COBID_SDO_RESPONSE()},

    {CO_KEY(0x1800, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)2},
    {CO_KEY(0x1800, 1, CO_UNSIGNED32|CO_OBJ_DN_R_), 0, CO_COBID_TPDO_DEFAULT(0)},
    {CO_KEY(0x1800, 2, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)254},
    {CO_KEY(0x1800, 3, CO_UNSIGNED16|CO_OBJ_D__RW), 0, (uintptr_t)0},
    {CO_KEY(0x1800, 5, CO_UNSIGNED16|CO_OBJ_D__RW), CO_TEVENT, (uintptr_t)10},  /* 1st trigger */

    {CO_KEY(0x1A00, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)3},
    {CO_KEY(0x1A00, 1, CO_UNSIGNED32|CO_OBJ_D__R_), 0, CO_LINK(0x2100, 0x01, 32)},
    {CO_KEY(0x1A00, 2, CO_UNSIGNED32|CO_OBJ_D__R_), 0, CO_LINK(0x2100, 0x02,  8)},
    {CO_KEY(0x1A00, 3, CO_UNSIGNED32|CO_OBJ_D__R_), 0, CO_LINK(0x2100, 0x03,  8)},

    {CO_KEY(0x2100, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)3},
    {CO_KEY(0x2100, 1, CO_UNSIGNED32|CO_OBJ___PR_), 0, (uintptr_t)&Obj2100_01_20},
    {CO_KEY(0x2100, 2, CO_UNSIGNED8 |CO_OBJ___PR_), 0, (uintptr_t)&Obj2100_02_08},
    {CO_KEY(0x2100, 3, CO_UNSIGNED8 |CO_OBJ___PR_), CO_TASYNC, (uintptr_t)&Obj2100_03_08},  /* 2nd trigger!! */

    CO_OBJ_DIR_ENDMARK  /* mark end of used objects */
};
```

**correct**:

```c
/* define the static object dictionary */
static struct CO_OBJ_T ClockOD[APP_OBJ_N] = {
    {CO_KEY(0x1000, 0, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)0},
    {CO_KEY(0x1001, 0, CO_UNSIGNED8 |CO_OBJ____R_), 0, (uintptr_t)&Obj1001_00_08},
    {CO_KEY(0x1005, 0, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)0x80},
    {CO_KEY(0x1017, 0, CO_UNSIGNED16|CO_OBJ_D__R_), 0, (uintptr_t)0},

    {CO_KEY(0x1018, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)4},
    {CO_KEY(0x1018, 1, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)0},
    {CO_KEY(0x1018, 2, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)0},
    {CO_KEY(0x1018, 3, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)0},
    {CO_KEY(0x1018, 4, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)0},

    {CO_KEY(0x1200, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)2},
    {CO_KEY(0x1200, 1, CO_UNSIGNED32|CO_OBJ_DN_R_), 0, CO_COBID_SDO_REQUEST()},
    {CO_KEY(0x1200, 2, CO_UNSIGNED32|CO_OBJ_DN_R_), 0, CO_COBID_SDO_RESPONSE()},

    {CO_KEY(0x1800, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)2},
    {CO_KEY(0x1800, 1, CO_UNSIGNED32|CO_OBJ_DN_R_), 0, CO_COBID_TPDO_DEFAULT(0)},
    {CO_KEY(0x1800, 2, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)254},
    {CO_KEY(0x1800, 3, CO_UNSIGNED16|CO_OBJ_D__RW), 0, (uintptr_t)0},
    {CO_KEY(0x1800, 5, CO_UNSIGNED16|CO_OBJ_D__RW), CO_TEVENT, (uintptr_t)10},  /* TPDO trigger */

    {CO_KEY(0x1A00, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)3},
    {CO_KEY(0x1A00, 1, CO_UNSIGNED32|CO_OBJ_D__R_), 0, CO_LINK(0x2100, 0x01, 32)},
    {CO_KEY(0x1A00, 2, CO_UNSIGNED32|CO_OBJ_D__R_), 0, CO_LINK(0x2100, 0x02,  8)},
    {CO_KEY(0x1A00, 3, CO_UNSIGNED32|CO_OBJ_D__R_), 0, CO_LINK(0x2100, 0x03,  8)},

    {CO_KEY(0x2100, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)3},
    {CO_KEY(0x2100, 1, CO_UNSIGNED32|CO_OBJ___PR_), 0, (uintptr_t)&Obj2100_01_20},
    {CO_KEY(0x2100, 2, CO_UNSIGNED8 |CO_OBJ___PR_), 0, (uintptr_t)&Obj2100_02_08},
    {CO_KEY(0x2100, 3, CO_UNSIGNED8 |CO_OBJ___PR_), 0, (uintptr_t)&Obj2100_03_08},

    CO_OBJ_DIR_ENDMARK  /* mark end of used objects */
};
```

NaserFaryad, 2022-01-31:
Thank you so so much
When I clear the `CO_TASYNC` from my TPDO the problem was solved!
now it works like a charm
[image]

michael-hillmann, 2022-01-31:
Great!! Have fun with the CANopen stack ;-)

*Btw: I change the title to reflect our outcome*

closed 2022-01-31 by michael-hillmann

farhadshabani, 2022-02-01:
Thanks everyone for the discussion! This seems extremely helpful! @michael-hillmann @NaserFaryad