embedded-office/canopen-stack #88: Accessing object 1014h:00h - SDO request success!

siddamsantosh, 2022-03-09:
**Describe the bug**
Accessing object 1014h:00h - SDO request success!

**To Reproduce**
Steps to reproduce the behavior:
1. Write value 00h to object 1014h:00h
2. send CAN request
	67F Tx SD  8  23 14 10 00 00 00 00 00
3. See error
	5FF Rx SD  8  60 14 10 00 00 00 00 00
	Accessing object 1014h:00h - SDO request success!

**Expected behavior**
Valid abort codes are 06090030h, 06090032h or 08000000h

**Screenshots**
[image]

**CANopen Stack (please complete the following information):**
 - Device: NXP S32K116
 - OS: none
 - Version v.4.1.7

**Additional context**
Add any other context about the problem here.

michael-hillmann, 2022-03-09:
Thanks for letting me know this issue. Could you add the relevant object directory entry line(s), please.

siddamsantosh, 2022-03-10:
Thanks Michael for your quick reply.
I think, these are the below object directory entry lines you are expecting. If not, kindly let me know some more details.

#define CO_TEMCYID ((const CO_OBJ_TYPE *)&COTEmcyId)  /*!< Dynamic COB-ID    */
...
...

const CO_OBJ_TYPE COTEmcyId = { 0, 0, 0, COTypeEmcyIdWrite };
...
...

/* 032 */ {CO_KEY(0x1014, 0, CO_UNSIGNED32|CO_OBJ__N_RW),      CO_TEMCYID, (uintptr_t)&para.com.indx_1014_0      },

commit 789f882a (2022-04-01): fix write to EMCY COB-ID (fix #88)
closed 2022-04-01 by michael-hillmann