embedded-office/canopen-stack #98: Is it possible to configure three different nodes with your stack on the same microcontroller?

guilleherreraf, 2022-04-13:
Hello again @michael-hillmann.
I was wondering if is it possible to create three instances of CO_NODE objects with three different OD that use the same CAN interface?
Thank you in advance for your time.

PS: I'm using the Portenta H7.

michael-hillmann, 2022-04-13:
Hmm, when thinking about this I came to the conclusion: it should be possible. My main issue in this (pure theoretical) thoughts is: "*How can we provide single CAN frames to all nodes (e.g. NMT commands, SYNCs etc)?*"

A first try could be:
- Split the CAN driver into two parts:
  - Low-Level CAN driver which
    - creates three CAN receive queues
    - Creates a single CAN sending queue
  - 3 very similar High-level CAN driver which
    - reads CAN frames out of **one** of the low-level CAN receive queues
    - write to the single CAN sending queue
- When setting up the CANopen nodes, with the High-level CAN driver selection, we essentially map the used receive queue from the Low-level CAN driver to the CANopen node

What do you think, is this a reasonable experiment for you?

michael-hillmann, 2022-04-13:
This is an image, what I tried to explain in words above. An image shows more than 100 words, right ;-)

[image]

guilleherreraf, 2022-04-13:
Thanks a lot for your answer Michael! I'll try to implement it and let you know how it goes!

dozack, 2022-07-08:
Hello, I am back! :)

This topic reminds me to refactor driver API to be "object-oriented", same as the higher layers. I was able to implement pretty primitive virtual bus in order to run N instances of canopen-stack on a single target. Driver was basically a list of callbacks and every time some of the nodes requested transfer, callbacks of every single listener except the transmitting one are called. In order to do that, i had to edit driver API for user to be able to pass custom void * reference to higher level handle.

michael-hillmann, 2023-09-12:
Hi, I guess your system is running fine. Closing this issue. In case of trouble, feel free to reopen it again.

closed 2023-09-12 by michael-hillmann