embedded-office/canopen-stack #99: publish driver

florenso, 2022-04-14:
Hi,

I wrote a driver for the esp32([espidf](https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/get-started/index.html)), and i would like to publish it.
but before i publish, i would like to see an other driver (not a dummy one)

is there a list with all drivers?

Its a great lib, works like a charm. I am going to use it as a master node, so i might contribute in the future ^^

michael-hillmann, 2022-04-17:
Hi, nice to hear that the project is useful for you.

Regarding the drivers, it is great that you want to contribute. Unfortunately I'm currently not maintaining a public repository containing drivers.

Maybe in the future, when I can invest the time for documenting and supporting all the hardware and cross-compiler complexities.

tpcorrea, 2022-05-12:
Hey, @florenso, I am using also the stack in an ESP32 (with Arduino framework) and I need a master node as well. We may join efforts for it.

gerald-hbs, 2022-08-08:
Hi,
I read these posts and I am intrested in the ESP32 driver too. I want to control my modell railroad with a canopen network. The ESP32 should fit my requirements. An old canopen stack is running. But this version have a dynamic OD. Thats nice. 
Can I download a test version of the driver and help to make it run?