embedded-office/canopen-stack #100: Timer setup

tpcorrea, 2022-05-12:
Hello, everyone! I am a bit confused with the two timer modes that can be used. In the cyclic mode, I must set a timer to generate periodic interrupts at a given period (say 1ms) and call the COTmrService() inside the interrupt service routine, correct? In this case, the timer driver employed must be the co_timer_swcycle.c. In this case, the reload parameter of the DrvTimerReload() is the number of ticks to the next event.

In the delta mode, do I need to call COTmrService() inside the ISR? What is the meaning of the reload parameter in this case?

Thank you very much in advance.

michael-hillmann, 2022-05-13:
Yes, you are right. In delta mode, you need to call COTmrService() in ISR, too.
The difference to the cyclic mode is, that your timer driver must reprogram the timer on every event to the exact time intervall for the upcoming needed timer event. In other words, we will get interrupts at every event time and no interrupts in between.

michael-hillmann, 2022-05-20:
The unit of the reload parameter depends on your driver. I typically configure the timer with a frequency of 1MHz. Then you get an accuracy of 1tick = 1us. The reload parameter is in ticks which is equal to 1us - very comfortable.