embedded-office/canopen-stack #101: SYNC messages

tpcorrea, 2022-05-12:
Dear Michael,

sorry to bother again but I have another question. According to https://canopen-stack.org/v4.2/usage/tpdo/#sync-producer, I only need to add objects 0x1005 and 0x1006 to the OD for generating the SYNC messages. I tried it here (I am using your code in a ESP32 node), but without success. Is there any other detail that I am missing?

Thanks a lot and kind regards,
Tomas

tpcorrea, 2022-05-12:
Hello, I tried a few things and the error seems to be that the COSync module initialization doesn't get the correct value from OD entry 0x1005, even if I put the 0x1005 and 0x1006 in the OD:

```
uint32_t obj1005 = CO_SYNC_COBID_ON | 0x80;    /* enable sync producer       */
uint32_t obj1006 = 1000000;                  /* sync cycle-time 1s in ms */

/* define the static object dictionary */
static struct CO_OBJ_T CnhCoNodeOD[APP_OBJ_N] = {
    {CO_KEY(0x1000, 0, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)0},
    {CO_KEY(0x1001, 0, CO_UNSIGNED8 |CO_OBJ____R_), 0, (uintptr_t)&Obj1001_00_08},
    {CO_KEY(0x1005, 0, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)0x80},
    {CO_KEY(0x1005, 0, CO_UNSIGNED32|CO_OBJ____RW), CO_TSYNCID, (uintptr_t)(&obj1005)},
    {CO_KEY(0x1006, 0, CO_UNSIGNED32|CO_OBJ____RW), CO_TSYNCCYCLE, (uintptr_t)(&obj1006)},
    {CO_KEY(0x1017, 0, CO_UNSIGNED16|CO_OBJ_D__R_), 0, (uintptr_t)0},
```

I only made it work by hardcoding the COB-ID inside the COSyncInit():

`    sync->CobId =  CO_SYNC_COBID_ON | 0x80;
`

Any idea why is this happening? Could it be endianness? 

BR,
Tomás

michael-hillmann, 2022-05-13:
Hi, not clear to me. Can you try the following configuration, please?

Maybe the sync producer is working with direct values only (by mistake):

```c
    {CO_KEY(0x1005, 0, CO_UNSIGNED32|CO_OBJ_D__RW), CO_TSYNCID, (uintptr_t)(CO_SYNC_COBID_ON | 0x80)},
    {CO_KEY(0x1006, 0, CO_UNSIGNED32|CO_OBJ_D__RW), CO_TSYNCCYCLE, (uintptr_t)(1000000)},
```

michael-hillmann, 2022-05-20:
Ok, then I guess we need to change the SYNC type functions slightly. In the SYNC type functions, the functions `COObjRdDirect`and `COObjWrDirect` are used, instead of `COObjRdValue` and `COObjWrValue`.

I will add a specific test to the test-suite and check the behavior.

tpcorrea, 2022-05-20:
> When you use normal process variables for OD entries, is this working?
> 
> ```
> uint32_t obj2000 = 0x12345678;
> 
> const struct CO_OBJ_T OD[] = {
>     :
>   {CO_KEY(0x2000, 0, CO_UNSIGNED32|CO_OBJ____R_), 0, (uintptr_t)&obj2000},
>   {CO_KEY(0x2001, 0, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)0x90abcdef},
>     :
> }
> ```
> 
> Can you read the content of these variables via SDO correctly?

I didn't try to read the OD entries via SDO, but if a trigger a TPDO to transmit the hard-coded and variable referenced signals, it fails. 
```
// Manufacturer Specific Objects
    {CO_KEY(0x2025, 0, CO_UNSIGNED8 |CO_OBJ___PR_), 0, (uintptr_t)&motor_temp},
    {CO_KEY(0x2026, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)2},
    {CO_KEY(0x2026, 1, CO_UNSIGNED8 |CO_OBJ___PR_), 0, (uintptr_t)0x02},
    {CO_KEY(0x2026, 2, CO_UNSIGNED8 |CO_OBJ___PR_), 0, (uintptr_t)&capacitor_temp},
```

Using only variable references works fine.
```
// Manufacturer Specific Objects
    {CO_KEY(0x2025, 0, CO_UNSIGNED8 |CO_OBJ___PR_), 0, (uintptr_t)&motor_temp},
    {CO_KEY(0x2026, 0, CO_UNSIGNED8 |CO_OBJ_D__R_), 0, (uintptr_t)2},
    {CO_KEY(0x2026, 1, CO_UNSIGNED8 |CO_OBJ___PR_), 0, (uintptr_t)&controller_temp},
    {CO_KEY(0x2026, 2, CO_UNSIGNED8 |CO_OBJ___PR_), 0, (uintptr_t)&capacitor_temp},
```

michael-hillmann, 2022-05-20:
Please keep an eye on the flags:

´´´c
{CO_KEY(0x2026, 1, CO_UNSIGNED8 |CO_OBJ___PR_), 0, (uintptr_t)0x02},
´´´

with your config, you will get the content of address 0x02. I guess your intension was to return the value 0x02.

Key for controlling this, is the Flag ´D´ in ´CO_OBJ___PR_´ vs. ´CO_OBJ_D_PR_´. See [Configuration/Object Entry](https://canopen-stack.org/v4.2/usage/configuration/#object-entry-key).

tpcorrea, 2022-05-20:
Alles klar, now it works fine, sorry for the mistake.

michael-hillmann, 2022-05-22:
You are welcome. Good to hear, that it now works.

closed 2022-05-22 by michael-hillmann