embedded-office/canopen-stack #102: OD entries must be ordered, otherwise stack may fail

tpcorrea, 2022-05-19:
**Describe the bug**
The stack works fine, but I have noticed that the Object Dictionary struct needs the entries to be ordered, i.e., from the smaller numbers to the larger ones.

**To Reproduce**
Just put an OD entry definition out of order and the stack may fail. In my case, I was defining first the TPDO parameters (0x1800) and mapping (0x1A00) and afterwards the RPDO parameters (0x1400) and mapping (0x1600). 

**Expected behavior**
It is not mandatory, but it is nice to have the possibility to include the OD entries out-of-order. If this feature will not be supported, please clearly state in the documentation the requirement of an ordered list. 

**CANopen Stack (please complete the following information):**
 - Device: ESP32 (ESP32-WROOM-32E
 - OS: Windows 10
 - Version v4.2

I am available for discussion or support if needed.

michael-hillmann, 2022-05-20:
Yes, this is intended behavior. The object dictionary in some projects becomes quite large, so we decide to keep it sorted to find an object entry as simple and fast as possible.

This requirement is documented in [Quickstert](https://canopen-stack.org/v4.2/start/quickstart/#mandatory-object-entries), maybe we place the same statement in the chapter [Object Dictionary](https://canopen-stack.org/v4.2/usage/dictionary/#object-dictionary), too?

tpcorrea, 2022-05-20:
Hi, Michael. Now I saw the information in Quickstart. Before submitting the issue, I checked the chapter OD, so I believe it would be nice if you add the same warning there. Thanks a lot!

michael-hillmann, 2022-05-23:
Ok, add an important note to website in configuration and usage to highlight this requirement.

closed 2022-05-23 by michael-hillmann