embedded-office/canopen-stack #107: Questions on event-timer and sync- based TPDOs

r-zan, 2022-06-16:
First of all, I would like to thank Michael Hillmann for the nice canopen-stack as well as the accompanying documentation. These have helped me to make progress in developing a CANopen servo drive, in which the canopen-stack is implemented (bare-metal) in a 32-bit microcontroller. I have reached the point where I am able to control the servo drive through SDO read/write and event-driven RPDO/TPDO based on mapped object changes. 

Now I intend to trigger the PDOs based on different mechanisms as the title of the issue suggests.

Event-timer-based TPDO
---------------------------
I have not been able to properly implement this. 

I did the following:
- Followed the timer driver implementation based on co_timer_dummy.c and co_timer_swcycle.c
- Set up a hardware timer ISR that is triggered every 1 msec.
- as explained in https://canopen-stack.org/v4.2/api/timer/, I put COTmrService(&(AppNode.Tmr)) in the hardware timer ISR and COTmrProcess(&(AppNode.Tmr)) inside the main while (1) loop
- Following https://canopen-stack.org/v4.2/examples/use-dynamic-object-dictionary/, I added a few objects through ODAddUpdate() and set the 3rd function parameter (*type_OD) as CO_TEVENT. These are the objects that are also mapped to a TPDO.
- Through ODCreateTPDOCom(), I set the type to 254, inhibit time to 100 and evtimer to 10.

Observation:
- The code gets stuck in the while(1) after looping for around 3 times.

Question:
what could have gone wrong here?

Sync-based TPDO
--------------------
What was done: 
- followed the example on SYNC consumer from https://canopen-stack.org/v4.2/usage/tpdo/
- Mapped the objects into the TPDO by using ODCreateTPdoMap(). The mapped objects' *type_OD (from the function ODAddUpdate) is set as 0.

Observation:
- The PDO is not being transmitted after SYNC messages are sent by the CANopen master for the number of times corresponding to what is defined in TPDO's subindex 2

Question:
What could have been missing in the implementation steps?

michael-hillmann, 2022-07-12:
Hmm, hard to say what is going on. Is there any error code stated in the node object? Sometimes there may be missing timers/actions which leads to this kind of behavior.

Maybe the tests can give you some additional hints:

- Event-timer-based TPDO: [`TS_TPdo_Tmr`](https://github.com/embedded-office/canopen-stack/blob/aeaba5b46d8ed8b6e1e2879414792f94dc26fe71/testsuite/tests/pdo_tx.c#L524)

- Sync-based TPDO: [`TS_TPdo_After3Sync`](https://github.com/embedded-office/canopen-stack/blob/aeaba5b46d8ed8b6e1e2879414792f94dc26fe71/testsuite/tests/pdo_tx.c#L229)

r-zan, 2022-07-15:
Thank you for the reply Michael. 
After more and more debugging, finally I have solved the event-timer-based TPDO issue and it came from the following:
- improper timer driver implementation
- In the while loop in Main(), there was a COObjWrValue to an object that is  set for an event-timer-based TPDO. This caused the code getting stuck during the execution. 
The sync-based TPDO is also working now, though I cannot ascertain where the problem really came from.
Once again, thank you!

closed 2022-07-15 by r-zan

michael-hillmann, 2022-07-15:
good to hear. have a nice day 👍