embedded-office/canopen-stack #110: Support for handling arbitrary object sizes in PDO mapping code

jernejsk, 2022-07-12:
**Is your feature request related to a problem? Please describe.**
I'm trying to implement CANopen application layer, which uses compound object with total size of 7 bytes. This same object must be PDO mapped. While object can be implemented by DOMAIN type and custom read and write handlers, PDO mapping code doesn't support any other size than 1, 2 and 4.

**Describe the solution you'd like**
I would like to add ability to map objects of arbitrary number of octets (up to 8, of course). I was thinking of extending code in CORPdoWrite() and COTPdoTx() functions with else block after standard lengths with COObjWrBufStart() and COObjRdBufStart(), respectively, for handling other object sizes. I would avoid using intermediary buffer and just directly write to object from CAN frame and vice versa.

**Describe alternatives you've considered**
None, because solution described above seems to me clean and easy to implement.

**Additional context**
Application layer specs are not written by us and we have to be compliant with them.

What is your opinion on this? I can open PR with suggested changes.

michael-hillmann, 2022-07-12:
What kind of data do you need to implement? Is it a manufacturer-specific structure, or a CANopen standard datatype, which is currently not supported (e.g. UNSIGNED56)?

In the case of a standard datatype: I would prefer to enhance the stack supporting INTEGER8,16,24,32,40,48,56,64 and UNSIGNED8,16,24,32,40,48,56,64.

In the case of a manufacturer-specific structure: is it something general (e.g. out of a device profile)?

michael-hillmann, 2022-07-12:
Using `COObjWrBufStart()` can get tricky, because the byte ordering on the CAN bus is fixed to LSB first while the `COObjWrBufStart()` reads the memory content bytewise in machine-dependent order. To get this work correctly in all cases, we could detect the machine-dependent order and improve the function `COObjRdValue()` to handle values with 1..8 bytes.

Some special considerations may apply to DSPs. In some DSP architectures, a single memory address holds more than 8 bits...

jernejsk, 2022-07-12:
It's not CiA profile, but actually CANopen ECSS (ECSS-E-ST-50-15C). It gives a lot of freedom, but time handling is pretty specific. Compound type for time object is specified as a struct with UNSIGNED24 and UNSIGNED32 fields. If I'm reading CiA301 correctly, compound types are meant to be accessed as a single object, so I guess support for UNSIGNED56 mapping would already be enough. Support for UNSIGNED64 mapping would be nice too, but for other purposes.

Can you suggest how to approach COObjRdValue() and COObjWrValue() improvements?

michael-hillmann, 2022-07-12:
I guess you want to implement `SCET`, right`?

What about the following idea:
```c
/* compound structure definition */
typedef struct SCET_T {
  uint32_t Coarse;
  uint32_t Fine;   /* lower 24bits used */
} SCET;

/* dedicated SCET set/get and other needed functions, for example: */
void setSCET(struct SCET_T *scet, uint32_t coarse, uint32_t fine)
{
  scet->Coarse = coarse;
  scet->Fine = fine;
}
```

Usage of SCET module:
```
/* variable allocation */
SCET mySCET = { 0 };

/* set SCET: */
setSCET(&mySCET, 1234L, 5678L);

/* storage in object dictionary (at sample index 2100h): */
const CO_OBJ DemoObjDir[] = {
    :
  /* SCET compound in a object with multiple sub-entries: */
  { CO_KEY(0x2100,0, CO_UNSIGNED8|CO_OBJ_D__R_), 0, (uintptr_t)2 },   /* highest subindex */
  { CO_KEY(0x2100,1, CO_UNSIGNED32|CO_OBJ___PRW), 0, (uintptr_t)&(mySCET.Coarse) },
  { CO_KEY(0x2100,2, CO_UNSIGNED32|CO_OBJ___PRW), 0, (uintptr_t)&(mySCET.Fine) },
    :
};
```

This way you can map the data elements individually to the PDO(s). Furthermore, you can access the data directly in the application (`mySCET.Course`) - however, using access functions is a better abstraction (but requires some code space).

michael-hillmann, 2022-07-12:
Implementing values up to 64bits in `COObjRdValue()/COObjWrValue()` is not an easy task. Especially considering all possible architectures. I have nothing prepared for this and must think about that some time ;-)

jernejsk, 2022-07-13:
Yes, it's SCET time, but using uint32_t for fine time wouldn't be according by specs. It would make structure one byte larger, so PDO sent by time master node, which is not made by us, would be rejected.

That's why I suggested using COObjWrBufStart() and COObjRdBufStart() - little code overhead because it's already used elsewhere and can handle arbitrary sizes. It would also be used only for non-standard sizes (other than 1, 2, 4), which requires custom handling anyway.

At very least, if you don't approve it as suitable solution, it seems to me to be good stop-gap measure for my case.

michael-hillmann, 2022-07-13:
Ahh, I see the weakness of the stack: according to the CiA standard, we can map the lower 24bits of a 32bit value into a PDO. Bitwise mapping is not possible right now. Is this something you may consider when I implement this possibility?

I think this would be a nice solution to your challenge:
```c
/* storage in object dictionary (at sample index 2100h): */
const CO_OBJ DemoObjDir[] = {
    :
  /* PDO mapping for PDO #0 */
  { CO_KEY(0x1A00,0, CO_UNSIGNED8|CO_OBJ_D__R_), 0, (uintptr_t)2 },                     /* highest subindex */
  { CO_KEY(0x1A00,1, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)CO_LINK(0x2100,1,24) }, /* 24bit of fine */
  { CO_KEY(0x1A00,2, CO_UNSIGNED32|CO_OBJ_D__R_), 0, (uintptr_t)CO_LINK(0x2100,2,32) }, /* 32bit of coarse */
    :
  /* SCET compound in a object with multiple sub-entries: */
  { CO_KEY(0x2100,0, CO_UNSIGNED8|CO_OBJ_D__R_), 0, (uintptr_t)2 },   /* highest subindex */
  { CO_KEY(0x2100,1, CO_UNSIGNED32|CO_OBJ___PRW), 0, (uintptr_t)&(mySCET.Coarse) },
  { CO_KEY(0x2100,2, CO_UNSIGNED32|CO_OBJ___PRW), 0, (uintptr_t)&(mySCET.Fine) },
    :
};
```

Well, I think using COObjWrBufStart() and COObjRdBufStart() is not suitable for common use because of the difference between *value* and *memory representation*. Nevertheless, if it is working on your architecture (you need to have a core working in little-endian) it is a possibility.

Let me have a look into the COTPdoTx() - maybe I can implement the bitwise mapping and additional place a callback function somewhere so you can use it for values greater than 32bits.

michael-hillmann, 2022-07-13:
Hi, I'm back again: bit-wise (1..64) mapping needs a lot of shifting and logic during each PDO transmission and reception. This increases code size a lot and therefore, I decide to enhance the byte-wise mapping to allow 1, 2, 3, or 4 bytes. In addition, if a single mapping uses more than 4 bytes, you can provide two callback functions which get the object and CAN message details to perform the corresponding read/write operation:

- For RPDO
```c
/*! \brief  PDO WRITE DATA CALLBACK
*
*    This function is called during PDO distribution of the PDO message
*    frame into the object dictionary when mapped data value consumes
*    more than 4 bytes.
*
* \param frm
*    Pointer to PDO message frame
*
* \param pos
*    The start index in the CAN message frame payload
*
* \param size
*    The mapped value size in bytes
*
* \param obj
*    Pointer to the target object entry
*/
void CORpdoWriteData(CO_IF_FRM *frm, uint8_t pos, uint8_t size, CO_OBJ *obj);
```

- For TPDO
```c
/*! \brief  PDO READ DATA CALLBACK
*
*    This function is called during PDO transmission when a mapped
*    data value consumes more than 4 bytes.
*
* \param frm
*    Pointer to PDO message frame
*
* \param pos
*    The start index in the CAN message frame payload
*
* \param size
*    The mapped value size in bytes
*
* \param obj
*    Pointer to the source object entry
*/
void COTpdoReadData(CO_IF_FRM *frm, uint8_t pos, uint8_t size, CO_OBJ *obj);
```

Now you may check the stack in the `develop` branch: here I pushed a version for you. With that version, you can check both solutions mentioned above.

michael-hillmann, 2022-07-13:
Just for your info: this enhancement is part of version 4.3.0.

commit e2663a06 (2022-07-13): document PDO mapping improvements (fix #110)