embedded-office/canopen-stack #116: Unable to change producer heartbeat time

silbstep, 2022-07-21:
Hi, firstly thank you for a great program with fantastic documentation. 

I have got the stack working very well on the ESP32 micro. I have one problem though and that is I am unable to change the time between heartbeats, I am not sure if this is my fault or if there is a bug in the program. 

I have created the variable uint16_t HBTime = 5000;
and then create this line in the Object Dictionary: ODAddUpdate(self, CO_KEY(0x1017, 0, CO_UNSIGNED16|CO_OBJ____RW), CO_THB_PROD, (uintptr_t)&HBTime);

I do get a heartbeat but it always seems to be about 1 second apart no matter what value I make HBTime. 

Your help or advice will be greatly appreciated.

michael-hillmann, 2022-07-22:
Hmm, this should work. Some ideas what you could check:

Have you had an error stored in the node object?
-> MyNode.Error (should be `CO_ERR_NONE`)

Maybe you are running out of software timers?
-> You can check this by calling `id = COTmrCreate(...)`; this should return an `id >= 0`.

Is your timer driver working as expected?
-> Just create some timer and check their timing

silbstep, 2022-07-22:
Thanks, Michael you immediately pointed me in the right direction and of course, the problem was due to my stupid error. The node object was alright, I did get CO_ERR_NON. I could successfully create a timer and I got an id of 1 which I assume means that I have one timer running. The problem was with the third option where the timing wasn't as expected. On further investigation, I found that somewhere (and quite a while back which is why I forgot about it) in the debugging process I was reloading the timer with a constant which of course then gave me a fixed time no matter what the value of HBTime. Thanks once again for your help.

closed 2022-07-22 by silbstep