embedded-office/canopen-stack #117: Heartbeat consumer event not executed if there are no heartbeats

jernejsk, 2022-08-04:
I'm implementing CAN bus redundancy management according to CANopen ECSS. There, CAN bus must be switched if predetermined number of heartbeats are missed. For this case, CONmtHbConsEvent() is useful, but only after first heartbeat is received from CAN. If bus is completely dead for some reason, this callback will never be executed. Quick workaround would be to inject appropriate HB in stack at boot, but I wonder if there is any better solution?

michael-hillmann, 2022-08-04:
For CiA301 this would break with a requirement from §7.5.2.19 "Monitoring of the heartbeat producer shall start after the reception of the first heartbeat." therefore, it is not suitable to integrate this into the common stack source.

I guess you are solving the potential situation during (hot) reset while system operates on redundant CAN bus #2, while the node starts with CAN bus #1 per default?

Maybe a small module with your algorithm to detect the active bus is suitable?

```c
typedef struct BUS_REDUNDANCY_T {
  CO_NODE *Node;
  uint32_t CycleTime;
  int16_t  ScanTmr;
  uint8_t  BusId;
} BUS_REDUNDANDY;

BUS_REDUNDANDY AppBus = {
  &AppNode,   /* pointer to parent CANopen node */
  50,         /* scan with 50ms listen phases */
  -1,         /* >=0: scanning is active */
  0           /* current active busId */
};

void AppBusStartScan(BUS_REDUNDANDY *bus)
{
  CO_NODE *node = bus->node;
  uint32_t ticks = COTmrGetTicks(node->Tmr, bus->CycleTime, CO_TMR_UNIT_1MS);
  uint32_t busId = 1 - bus->BusId;
  int16_t  bus->ScanTmr = COTmrCreate(node->Tmr, ticks, 0, AppBusTimeout, &busId);

  /* function to switch physical bus to 'busId' */

  bus->busId = busId;
}

void AppBusTimeout(void *args)
{
    AppBusStartScan((BUS_REDUNDANDY *)args);
}

void AppBusStopScan(BUS_REDUNDANDY *bus)
{
  if (bus->ScanTmr >= 0) {
    COTmrDelete(node->Tmr, bus->ScanTmr);
  }
  bus->ScanTmr = -1;
}
```

using during your startup:

```c
void foo()
{
  CONodeInit(&AppNode, AppSpec);
    :
  CONodeStart(&Clk);
  AppBusStartScan(&AppBus);
}
```

and in the callback:

```c
void CONmtHbConsChange(CO_NMT *nmt, uint8_t nodeId, CO_MODE mode)
{
  AppBusStopScan(&AppBus);
}
```

Just an idea (without any error checking) ;-)

jernejsk, 2022-08-05:
Ok, thanks for the info and idea.

closed 2022-08-05 by jernejsk