embedded-office/canopen-stack #118: TPDO transmission type configuration from client side

r-zan, 2022-08-15:
Hello,

I am testing the configurability of TPDO transmission type from client side (through SDO write). The test cases and findings are as follow:

- Change from default transmission type of SYNC-based to timer-based
  
   On the server side, the default is configured through following in the firmware:
    
   ODCreateTPDOCom(self, 1, CO_COBID_TPDO_DEFAULT(1), 2, 1000, 0);   
   ODAddUpdate(self, CO_KEY(0x60F4, 0, CO_SIGNED32|CO_OBJ___PRW), 0, (uintptr_t)&obj1);
   uint32_t TPDO_map_1[2];
   TPDO_map_1[0] = CO_LINK(0x60F4, 0, 32);
   ODCreateTPdoMap(self, 1, TPDO_map_1, 1);

   As such, from the client side, the TPDO is transmitted simply by sending 2x SYNC signals.
   Now, the transmission type is changed to timer-based through the following SDO write commands:

   SDO write to object (index: 0x1801, subidx: 2), with value: 0xFE
   SDO write to object (index: 0x1801, subidx: 5), with value: 1000

   **The TPDO is transmitted every 1000 msec. Just as expected.**

- Change from default transmission type of SYNC-based to object-change-based
   
   The configuration on the server side is the same as the above.

    The transmission type is changed to timer-based through the following SDO write commands:
   SDO write to object (index: 0x1801, subidx: 2), with value: 0xFE
   SDO write to object (index: 0x1801, subidx: 5), with value: 0

   **The TPDO is not transmitted at all even by externally stimulating the change of object value.**

- Change from default transmission type of object-change-based to timer-based

   On the server side, the default is configured through following in the firmware:
    
   ODCreateTPDOCom(self, 1, CO_COBID_TPDO_DEFAULT(1), 0xFE, 1000, 0);   
   ODAddUpdate(self, CO_KEY(0x60F4, 0, CO_SIGNED32|CO_OBJ___PRW), CO_TASYNC, (uintptr_t)&obj1);
   uint32_t TPDO_map_1[2];
   TPDO_map_1[0] = CO_LINK(0x60F4, 0, 32);
   ODCreateTPdoMap(self, 1, TPDO_map_1, 1);

   As such, the TPDO is transmitted simply by externally stimulating the change of object value.
 
   Now, the transmission type is changed to timer-based through the following SDO write commands:
   
   SDO write to object (index: 0x1801, subidx: 5), with value: 1000
   
    **The TPDO is transmitted not only every 1000 msec., but also through object change.** 

Question: how can we properly configure the TPDO transmission type from client side, so that it would work only through object change?

michael-hillmann, 2022-08-15:
First, thanks for your notification. I remember such a behavior in v4.3.1. Could you re-try your scenarios with the current development version in branch `develop-type-modules`, please?

*Note: there is a breaking change in defining the object dictionary and some structural changes. I try to release this version until end of September. I'm in testing phase -  so great, that you came up with some additional test cases out of your practise.*

michael-hillmann, 2022-08-15:
The intension of this asynchronous behavior is:
- when an object entry is defined with `CO_TASYNC` (or with new `A` flag: `CO_OBJ___APRW`), a transmission of all related PDOs are triggered (depending on PDO mapping) when the object value is changed with an API function.

- the PDO specific inhibit timer is respected after each transmission of the PDO.

- when an event timer (subindex 5 in communication) is active - the event timer triggered transmission is (re-)synchronized on this immediate object change transmission. With this behavior, we can ensure that the object change is transmitted as soon as possible and without a change we define the minimum transmission cycle time with the event timer.

- when working with synchronized PDOs the object change method is not that useful, because we should transmit on received Sync messages only. Therefore we shouldn't define the object entry with `CO_TASYNC` (respectivly `CO_OBJ___APRW`).

michael-hillmann, 2022-08-15:
Due to the fact, that the ASYNC mode is manufacturer specific, we have no simple way of changing this object property via SDO transfers. As I see in your tests, you are using dynamic object entries.

Good news: there could be a possible solution, because the object entry definition is in RAM. We just need to define the communication event which changes the type `CO_TASYNC` to `0` (or `0` to `CO_TASYNC`).

michael-hillmann, 2022-08-15:
**Question: how can we properly configure the TPDO transmission type from client side, so that it would work only through object change?**

Read your question again and can only suggest:

On the server side, the default is configured through following in the firmware:

```c
ODCreateTPDOCom(self, 1, CO_COBID_TPDO_DEFAULT(1), 0xFE, 0, 0);
ODAddUpdate(self, CO_KEY(0x60F4, 0, CO_SIGNED32|CO_OBJ___PRW), CO_TASYNC, (uintptr_t)&obj1);
uint32_t TPDO_map_1[2];
TPDO_map_1[0] = CO_LINK(0x60F4, 0, 32);
ODCreateTPdoMap(self, 1, TPDO_map_1, 1);
```

**And keep the object (index: 0x1801, subidx: 5), at value: 0.**

r-zan, 2022-08-16:
Hello Michael,

Thank you for your quick replies, it is much appreciated!

As you suggested, I followed the default configuration at the server side.

Additionally, in the server side, I periodically check for the event timer (index: 0x1801, subidx: 5), if it is set to a value >0 from the client side, it will then execute (based on the possible solution that you proposed): 

ODAddUpdate(&VLC_CO, CO_KEY(0x60F4, 0, CO_SIGNED32|CO_OBJ___PRW), 0, (uintptr_t)&obj1);

However, this didn't help to change the transmission type to be fully event-timer-based. I still see the same behavior that the transmission type is both event-timer-based and object-change-based. 

Is there a step that I could have been missing in the implementation?

michael-hillmann, 2022-08-16:
Hmm, I guess you are performing the "re-configuration" of the PDO communication setup during `OPERATIONAL` mode? If possible, try to limit such changes to `PRE-OPERATIONAL` mode.

Anyway: is there a different behavior after requesting a communication reset via NMT?

r-zan, 2022-08-16:
Given the periodic check and execution of ODAddUpdate I mentioned before, test cases and findings are as follow (for each test case the CANopen slave firmware execution is reset):

- case 1: 
   
   client: NMT set to operational
   server: TPDO transmission based on object-change
   client: SDO write to object (index: 0x1801, subidx: 5), with value: 1000
   server: TPDO transmission based on object-change and timer-event
   client: NMT set to preoperational
   server: TPDO transmission is still happening
   client: NMT set to reset communication
    server: TPDO transmission is still happening

- case 2:
   
   client: NMT set to preoperational
   client: SDO write to object (index: 0x1801, subidx: 5), with value: 1000
   lient: NMT set to operational
   server: no TPDO transmission

- case 3:

   client: NMT set to preoperational
   client: SDO write to object (index: 0x1801, subidx: 5), with value: 1000
   client: NMT reset communication
   client: NMT set to operational
   server: no TPDO transmission

michael-hillmann, 2022-08-16:
I have written some tests in the testsuite, corresponding to your list. It is working (even the change of object properties while working in operational mode):

```c
/*------------------------------------------------------------------------------------------------*/
/*
*          This testcase will check issue #118 (change object entry property)
*/
/*------------------------------------------------------------------------------------------------*/
TS_DEF_MAIN(TS_TPdo_ChangeAsyncProperty)
{
    CO_IF_FRM frm;
    CO_NODE   node;
    uint32_t  tpdo_id      = 0x40000180;
    uint32_t  tpdo_map     = 0x25000B08;
    uint8_t   tpdo_type    = 254;
    uint16_t  tpdo_inhibit = 0;
    uint16_t  tpdo_evtime  = 0;
    uint8_t   tpdo_len     = 1;
    uint8_t   data8        = 0x91;

    TS_CreateMandatoryDir();
    TS_CreateTPdoCom(1, &tpdo_id, &tpdo_type, &tpdo_inhibit, &tpdo_evtime);
    TS_CreateTPdoMap(1, &tpdo_map, &tpdo_len);
    TS_ODAdd(CO_KEY(0x2500, 0x0B, CO_UNSIGNED8|CO_OBJ___PRW), CO_TASYNC, (CO_DATA)(&data8));
    TS_CreateNode(&node,0);
    
    TS_NMT_SEND(1, 1);                                       /* set node-id 1 to operational      */

    TS_SDO_SEND(0x2B, 0x1801, 5, 1000);                      /* write to 1801:05, value 1000      */
    CHK_SDO0_OK(0x1801, 5);                                  /* check for success                 */

                                                             /* change object entry property -A   */
    TS_ODAdd(CO_KEY(0x2500, 0x0B, CO_UNSIGNED8|CO_OBJ___PRW), 0, (CO_DATA)(&data8));

    TS_Wait(&node, 1100);                                    /* wait for 1000ms +10%              */
    CHK_CAN(&frm);                                           /* check for a CAN frame             */

    CODictWrByte(&node.Dict, CO_DEV(0x2500, 0x0B), 0x22);    /* change object value               */
    CHK_NOCAN(&frm);                                         /* check for no CAN frame            */

    TS_SDO_SEND(0x2B, 0x1801, 5, 0);                         /* write to 1801:05, value 0         */
    CHK_SDO0_OK(0x1801, 5);                                  /* check for success                 */

                                                             /* change object entry property +A   */
    TS_ODAdd(CO_KEY(0x2500, 0x0B, CO_UNSIGNED8|CO_OBJ___PRW), CO_TASYNC, (CO_DATA)(&data8));

    TS_Wait(&node, 1100);                                    /* wait for 1000ms +10%              */
    CHK_NOCAN(&frm);                                         /* check for no timer CAN frame      */

    CODictWrByte(&node.Dict, CO_DEV(0x2500, 0x0B), 0x33);    /* change object value               */
    CHK_CAN(&frm);                                           /* check for a CAN frame             */

    CHK_NO_ERR(&node);                                       /* check error free stack execution  */
}
```

michael-hillmann, 2022-08-16:
```c
/*------------------------------------------------------------------------------------------------*/
/*
*          This testcase will check issue #118 (case 1)
*/
/*------------------------------------------------------------------------------------------------*/
TS_DEF_MAIN(TS_TPdo_ChangeEventTime)
{
    CO_IF_FRM frm;
    CO_NODE   node;
    uint32_t  tpdo_id      = 0x40000180;
    uint32_t  tpdo_map     = 0x25000B08;
    uint8_t   tpdo_type    = 254;
    uint16_t  tpdo_inhibit = 0;
    uint16_t  tpdo_evtime  = 0;
    uint8_t   tpdo_len     = 1;
    uint8_t   data8        = 0x91;

    TS_CreateMandatoryDir();
    TS_CreateTPdoCom(1, &tpdo_id, &tpdo_type, &tpdo_inhibit, &tpdo_evtime);
    TS_CreateTPdoMap(1, &tpdo_map, &tpdo_len);
    TS_ODAdd(CO_KEY(0x2500, 0x0B, CO_UNSIGNED8|CO_OBJ___PRW), CO_TASYNC, (CO_DATA)(&data8));
    TS_CreateNode(&node,0);
    
    TS_NMT_SEND(1, 1);                                       /* set node-id 1 to operational      */

    CODictWrByte(&node.Dict, CO_DEV(0x2500, 0x0B), 0x11);    /* change object value               */
    CHK_CAN  (&frm);                                         /* check for a CAN frame             */

    TS_SDO_SEND(0x2B, 0x1801, 5, 1000);                      /* write to 1801:05, value 1000      */
    CHK_SDO0_OK(0x1801, 5);                                  /* check for success                 */

    TS_Wait(&node, 1100);                                    /* wait for 1000ms +10%              */
    CHK_CAN(&frm);                                           /* check for a CAN frame             */

    TS_NMT_SEND(128, 1);                                     /* set node-id 1 to pre-operational  */

    CODictWrByte(&node.Dict, CO_DEV(0x2500, 0x0B), 0x22);    /* change object value               */
    CHK_NOCAN(&frm)                                          /* check for no CAN frame            */

    TS_NMT_SEND(130, 1);                                     /* communication reset on Node 1     */
    CHK_CAN(&frm);                                           /* check for a CAN frame (bootup)    */
    CHK_BOOTUP(frm, 1);                                      /* BootUp message of Node-ID #1      */

    CODictWrByte(&node.Dict, CO_DEV(0x2500, 0x0B), 0x33);    /* change object value               */
    CHK_NOCAN(&frm)                                          /* check for no CAN frame            */

    TS_Wait(&node, 1100);                                    /* wait for 1000ms +10%              */
    CHK_NOCAN(&frm);                                         /* check for no CAN frame            */

    CHK_NO_ERR(&node);                                       /* check error free stack execution  */
}
```

michael-hillmann, 2022-08-16:
```c
/*------------------------------------------------------------------------------------------------*/
/*
*          This testcase will check issue #118 (case 2)
*/
/*------------------------------------------------------------------------------------------------*/
TS_DEF_MAIN(TS_TPdo_SetEventTime)
{
    CO_IF_FRM frm;
    CO_NODE   node;
    uint32_t  tpdo_id      = 0x40000180;
    uint32_t  tpdo_map     = 0x25000B08;
    uint8_t   tpdo_type    = 254;
    uint16_t  tpdo_inhibit = 0;
    uint16_t  tpdo_evtime  = 0;
    uint8_t   tpdo_len     = 1;
    uint8_t   data8        = 0x91;

    TS_CreateMandatoryDir();
    TS_CreateTPdoCom(1, &tpdo_id, &tpdo_type, &tpdo_inhibit, &tpdo_evtime);
    TS_CreateTPdoMap(1, &tpdo_map, &tpdo_len);
    TS_ODAdd(CO_KEY(0x2500, 0x0B, CO_UNSIGNED8|CO_OBJ___PRW), CO_TASYNC, (CO_DATA)(&data8));
    TS_CreateNode(&node,0);
    
    TS_NMT_SEND(128, 1);                                     /* set node-id 1 to pre-operational  */

    TS_SDO_SEND(0x2B, 0x1801, 5, 1000);                      /* write to 1801:05, value 1000      */
    CHK_SDO0_OK(0x1801, 5);                                  /* check for success                 */

    TS_NMT_SEND(1, 1);                                       /* set node-id 1 to operational      */

    TS_Wait(&node, 1100);                                    /* wait for 1000ms +10%              */
    CHK_CAN(&frm);                                           /* check for a CAN frame             */

    CODictWrByte(&node.Dict, CO_DEV(0x2500, 0x0B), 0x22);    /* change object value               */
    CHK_CAN(&frm)                                            /* check for a CAN frame             */

    CHK_NO_ERR(&node);                                       /* check error free stack execution  */
}
```

michael-hillmann, 2022-08-16:
and

```c
/*------------------------------------------------------------------------------------------------*/
/*
*          This testcase will check issue #118 (case 3)
*/
/*------------------------------------------------------------------------------------------------*/
TS_DEF_MAIN(TS_TPdo_SetEventTimeAndReset)
{
    CO_IF_FRM frm;
    CO_NODE   node;
    uint32_t  tpdo_id      = 0x40000180;
    uint32_t  tpdo_map     = 0x25000B08;
    uint8_t   tpdo_type    = 254;
    uint16_t  tpdo_inhibit = 0;
    uint16_t  tpdo_evtime  = 0;
    uint8_t   tpdo_len     = 1;
    uint8_t   data8        = 0x91;

    TS_CreateMandatoryDir();
    TS_CreateTPdoCom(1, &tpdo_id, &tpdo_type, &tpdo_inhibit, &tpdo_evtime);
    TS_CreateTPdoMap(1, &tpdo_map, &tpdo_len);
    TS_ODAdd(CO_KEY(0x2500, 0x0B, CO_UNSIGNED8|CO_OBJ___PRW), CO_TASYNC, (CO_DATA)(&data8));
    TS_CreateNode(&node,0);
    
    TS_NMT_SEND(128, 1);                                     /* set node-id 1 to pre-operational  */

    TS_SDO_SEND(0x2B, 0x1801, 5, 1000);                      /* write to 1801:05, value 1000      */
    CHK_SDO0_OK(0x1801, 5);                                  /* check for success                 */

    TS_NMT_SEND(130, 1);                                     /* reset communication node-id 1     */
    CHK_CAN(&frm);                                           /* check for a CAN frame (bootup)    */

    TS_NMT_SEND(1, 1);                                       /* set node-id 1 to operational      */

    TS_Wait(&node, 1100);                                    /* wait for 1000ms +10%              */
    CHK_CAN(&frm);                                           /* check for a CAN frame             */

    CODictWrByte(&node.Dict, CO_DEV(0x2500, 0x0B), 0x22);    /* change object value               */
    CHK_CAN(&frm)                                            /* check for a CAN frame             */

    CHK_NO_ERR(&node);                                       /* check error free stack execution  */
}
```

michael-hillmann, 2022-08-16:
Btw. all tests are passed.

Maybe you should look to `AppNode->Error`: is there an error stored?

r-zan, 2022-08-17:
Thank you for your replies @michael-hillmann , I will put an update here after performing some tests.

r-zan, 2022-08-18:
I performed the tests following the steps in the testcases you presented above, using my CANopen client software. For the first 2 testcases, the results are as follow:
- https://github.com/embedded-office/canopen-stack/issues/118#issuecomment-1216589748

TS_ODAdd is implemented in the server's code. The function is automatically executed based on event timer value that is newly assigned.

In this part:
    CODictWrByte(&node.Dict, CO_DEV(0x2500, 0x0B), 0x22);    /* change object value               */
    CHK_NOCAN(&frm);                                         /* check for no CAN frame            */
There was actually a CAN frame.

I followed all the steps until the end and checked AppNode->Error, there was no error.

- https://github.com/embedded-office/canopen-stack/issues/118#issuecomment-1216590162

All the checks were passed, however at the end of the test I checked for AppNode->Error, it showed CO_ERR_SYNC_RES

michael-hillmann, 2022-08-18:
The error `CO_ERR_SYNC_RES` gives us a hint into the direction of SYNC Producer - but there is no SYNC producer configured in the tests!? Sounds very strange...

One idea to check if you are you working with an RTOS:
- Check that the stack sizes of all tasks are large enougth.
- If possible, enable the MPU to protect the dynamic object dictionary.

r-zan, 2022-08-19:
I will get back here after some more testing. As a matter of fact, no RTOS is used.

r-zan, 2022-08-23:
> The error `CO_ERR_SYNC_RES` gives us a hint into the direction of SYNC Producer - but there is no SYNC producer configured in the tests!? Sounds very strange...
>
> One idea to check if you are you [...]

I defined object 0x1005 in my implementation, does that mean that SYNC producer is somehow configured?

I've just performed the 2nd test I did before https://github.com/embedded-office/canopen-stack/issues/118#issuecomment-1219206567 , but this time by not defining object  0x1005. At the end of the test, AppNode->Error showed CO_ERR_OBJ_NOT_FOUND . Could I be missing/misconfiguring anything ?

closed 2023-10-26 by r-zan