embedded-office/canopen-stack #124: Not able to find object dictionary file in the stack and no idea which tool need to use to generate the EDS file

ankit-eic, 2022-08-25:
Hello,

I need some information for the object dictionary and EDS file which is required when we are running the conformance test tool to pass the CANOpen stack.

So, I cannot find the object dictionary file for the CANOpen stack in this stack and do not know which tool to use to generate the EDS file for a node to pass the conformance test tool.

Could anyone please give some insight on the above queries?

Thanks,
Ankit

ankit-eic, 2022-08-29:
Hi,

Could I get any update on the above question? Thanks in advance.

Regards,
Ankit

ankit-eic, 2022-08-31:
Hello, Thanks for the clarification on the object dictionary part. How I can get to know about the commercial CANopen stack?

michael-hillmann, 2022-09-01:
Hi, there is no commercial CANopen stack. You can use this stack for free and may buy some assistance for your team to focus your project workload on the application.

closed 2022-09-06 by michael-hillmann