embedded-office/canopen-stack #130: Failed to link files

Moadib129, 2022-11-25:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. setup object entry to '...'
2. send CAN request '....'
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**CANopen Stack (please complete the following information):**
 - Device: [e.g. STM32F429]
 - OS: [e.g. none]
 - Version [e.g. v1.2.3]

**Additional context**
Add any other context about the problem here.

Moadib129, 2022-11-25:
Thank You for looking!
I have downloaded the repository and run the commands given in the end of readme file.
The first command runs great but when it time to run second command:

michael-hillmann, 2023-09-08:
Sorry for the late response. Have you solved this issue?

I have no glue, what is going wrong at your side.
Is the cmake configuration running without errors?
On which Host operating system are you working?