embedded-office/canopen-stack #131: Heartbeat producer issue - HB frame not transmitting

KFatyga, 2022-12-09:
Hello,
I have an issue with the node not producing heartbeat, while using the example ClockApp. I'm running it on an embedded system, based on TI F28379d Delfino MCU. I've developed basic drivers which worked with previous version (4.1) about a year ago, and the HB was operating.
I upgraded to the current version, and lost the heartbeat entirely. There is only a singular HB frame with value zero transmitted once at the start of the program. Any changes in HB entry in OD did not help. The node is responding to Sync frame, transmits PDOs synchronously. The clock from the ClockApp also works fine. Only the HB is missing.
My testbench is the TI F28379D Delfino Launchpad and Peak Can sniffer.
Would You have any suggestions as to where to look for the solution?

michael-hillmann, 2023-09-08:
Sorry for the late response. Have you solved this issue?

As a first idea, I guess the breaking change for version 4.4.0 may cause this issue. You can check the required migration steps, stated in the release notes.

KFatyga, 2023-09-11:
Hello,
I've checked for the change you mentioned - everything seems fine - both the underscore and CO_T prefix are included in the changed version.
The OD entry for the heartbeat:
`{CO_KEY(0x1017, 0, CO_OBJ_____RW), CO_THB_PROD,    (CO_DATA)(&Obj1017_00_10)},`
with the data object:
`static uint16_t Obj1017_00_10 = 500;`
The result:
[image]
The HB is transmitted once on startup with zero value - which, if I remember correctly is a desired behaviour - but it is not followed by the HB afterwards

KFatyga, 2023-09-12:
Following Your instructions, I checked: indeed, the `node->Nmt.Tmr` was set to `-1`,
Breakpoint in `COTNmtHbProdInit()` shows that the value read in line 148: `(void)uint16->Read(obj, node, &cycTime, sizeof(cycTime));` is read as zero. I forced the value to 500, works just fine, so I assume the issue is somewhere on read OD value level?

michael-hillmann, 2023-09-12:
Ok, this is an issue related to DSP datatypes. I will talk to my DSP specialist and come back with a solution.

commit e9cbe020 (2023-09-12): use sizeof for DSP 16bit datatype (fix #131)
closed 2023-10-20 by KFatyga