embedded-office/canopen-stack #155: How to: master and more than default number of PDOs

mullinsen, 2023-12-13:
Hi, I love this project but I’m new to CANopen and struggling to figure out a couple of things.

I found a question from 2022 where someone wanted to use CanOpen Stack for a NMT master node, and the reponse was that CanOpen Stack is intended as «NMT Slave only», but working towards supporting master nodes. Is this still the case? If it now supports both, how to I configure a node to be NMT master?

Second question, I need more than 4 RPDOs in a slave node. However, even though they are all defined in the OD, CanOpen Stack does not seem to process more than the 4 first RPDOs when initializing. How do I configure more than 4 RPDOs/TPDOs?

andrewteta, 2024-01-04:
I'm also interested in implementing a NMT master node. Any help would be much appreciated!

mullinsen, 2024-01-04:
@andrewteta I actually found the answers I was looking for in the following closed issues:
Multiple PDOs: https://github.com/embedded-office/canopen-stack/issues/60
NMT master (not supported, but workaround suggested was sufficient for me) : https://github.com/embedded-office/canopen-stack/issues/97

closed 2024-01-04 by mullinsen